volumio/Volumio2 #1754: dtoverlays gpio-shutdown and gpio-poweroff do not work on Raspberry Pi

amanita-citrina, 2019-04-26:
dtoverlays for shutdown and poweroff seem to have no effect.
I added the following lines to /boot/config.txt:
`dtoverlay=gpio-shutdown,gpio_pin=6,active_low=1`
`dtoverlay-gpio-poweroff,gpiopin=5,active_low=1`

Using `gpio readall` showed GPIO 6 as an input with value=1. When connecting it to GND, the value changed to 0 as expected, but no shutdown occurred.

The dtoverlays worked on another distribution (moode audio).

System configuration:
- Raspberry Pi 3B
- HifiBerry DAC+ Pro
- Volumio 2.575 (also tried some earlier versions)

dubaleeiro, 2019-05-04:
The HifiBerry DAC+ Pro uses the GPIOs 5 & 6:
[https://www.hifiberry.com/build/documentation/gpio-usage-of-hifiberry-boards/]
I would suggest that you change to other pins.

amanita-citrina, 2019-05-04:
Actually, the HifiBerry **Digi+** Pro uses GPIOs 5 and 6. On the **DAC+** Pro, these GPIOs are unused. As I was able to use the dt_overlays with moode audio, this seems indeed a volumio issue.

dubaleeiro, 2019-05-05:
You are right. I have now tested the 'gpio-shudown' overlay with pin 17 in a raspberry pi zero w with volumio and it did not work either. I am using JustBoom Zero AMP.

Jest0r, 2019-05-06:
Same problem here with volumio2, Allo BOSS 1.2 + Allo Isolator 1.2. Tried with GPIO 26 and 23

Jest0r, 2019-05-17:
Works for me now. The problem was in my case that the Isolator hat doesn't propagate the GPIO pins properly as input. Works nicely with a GPIO extension board in between (or without the Isolator)

dubaleeiro, 2019-05-17:
Good news! Could you plz tell which pin are you using?