volumio/Volumio2 #1757: Better device identification

ashthespy, 2019-05-16:
Currently devices are identified via the `Hardware` tab of `/proc/cpuinfo` which is a bit ambiguous. Wouldn't it be better to use `/proc/device-tree/name` instead?

volumio, 2019-05-17:
Not a bad idea at all! We need however to test with all the devices we do support...

ashthespy, 2019-05-17:
An alternative would be to just add the device info during build time. We could sneak it into `os-info` for example.

gkkpch, 2019-12-14:
I agree, cpuinfo is not very reliable because it depends on what board developers/ suppliers have put in the device tree source.
But there is already a build option in the kernel command line parameters which I have used for different devices. The parameter is "hwdevice=<value>".
So, read "/proc/cmdline" and parse "hwdevice"

ashthespy, 2020-02-03:
Was just revisiting this, a quick grep shows that only a few boards currently use `hwdevice`, and is again set manually for each board. 
```bash
scripts\odroidn2config.sh
hwdevice=Odroid-N2 

scripts\odroidxu4config.sh
hwdevice=OdroidXU4 

scripts\rock64config.sh
hwdevice=Rock64

scripts\boards\odroidn2config.sh
hwdevice=Odroid-N2 

scripts\boards\odroidxu4config.sh
hwdevice=OdroidXU4

scripts\boards\rock64config.sh
hwdevice=Rock64
```
Do you recommend that we define our own naming convention viz  `/proc/device-tree/model` which normally is more comprehensive?

gkkpch, 2020-02-04:
Problem is that only with mainline kernels from the last two or three years, people are paying attention to supplying a proper hw model using the device tree. Our hwdevice parameter helps to avoid issues in cases where /proc/cpuinfo does not return correct info (as with numerous pre 4.0 kernels)
Cheers - Gé

ashthespy, 2020-02-04:
@gkkpch Fair enough! 
Then what would be the way forward? Should we then define a list of `hwdevice`'s in a central location (maybe over at `volumio/build`) and add them to the `system` plugin folder during build time?

ashthespy, 2022-01-22:
Repo (but not old issues) has been moved over to https://github.com/volumio/volumio3-backend

closed 2022-01-22 by ashthespy