volumio/Volumio2 #1761: Defaulting to caps for first letter in UI for Network Shares (why)

DarkBlaez, 2019-05-29:
Why has this bug in the UI remained. If this was tested completely it would have been caught. In the box for NAS IP or Name you are capitalizing the first letter, please do not do this, you make name resolution somewhat problematic. Also, in the Path field you do this again and also in the Advance section for User. Please do not do this, let the user decide what they enter here. Otherwise we do not know what is taking place behind the scenes. Not sure how this was even passing testing. Please fix and update

Thanks
DB

volumio, 2019-05-29:
We are not capitalizing. Are you sure it's not your browser doing this?

closed 2019-05-29 by DarkBlaez

DarkBlaez, 2019-05-29:
Damn... you are right it's the dame Safari browser on the iPad, thanks