volumio/Volumio2 #1762: M3U files

mashby05, 2019-05-29:
Hello

I have volumio set up on a Raspberry pi b+. I have a lot of M3U playlists that link to music on a NAS.I want to use them with volumio. I have got them to show up in playlists by copping them to data/playlist. I have tried issue #1722 and m3u2volumio but since i am quite new i was not able to get ether to work. Is anyone able to give me a very in depth tutorial/instruction sequence to get volumio working with m3u playlists or how to convert them to the .json playlist format.

Thanks Matt

azathoth123, 2019-05-29:
Hi Matt,

I had success by updating to the latest version and replacing the file with
the updated one from the commit
https://github.com/SteveTalbot/Volumio2/commit/dfdb70270bb21fd77fb3d2b64e6933a24915a43a
If you can log in via ssh to the raspberrypi, then replace the file
app/plugins/music_service/mpd/index.js
<https://github.com/SteveTalbot/Volumio2/commit/dfdb70270bb21fd77fb3d2b64e6933a24915a43a#diff-2965d5982753ac5ebe2a4ba4f826b3a2>
with
the patched version.

I've been working this way for a few months now without issue, where i have
the music files in the same folder as the m3u file.

Steven

mashby05, 2019-05-30:
Hi steven

i have tried changing the index.js file but sill no luck. It wont show the music in the playlist. The music is on a nas and the playlists are absolute.

is there a easy way to convert the playlists to the correct format?
Thanks Matt

azathoth123, 2019-05-31:
Make sure you restart the Pi.
I can't see the contents of the m3u either, only the tracks after they are
added to the current playlist after clicking on the m3u.
You may need to use the same release that the change is based on in the
case that the current release has moved ahead and is no longer compatible,
it was some time ago that I did this.
You could also check that the contents of the m3u are resolvable from the
pi (from the terminal).

Otherwise there is a script somewhere that converts the m3u...

Steven.

mashby05, 2019-06-06:
Hi steven,
i will try to reinstall and change the file again but i was wondering if using relative or absolute paths would make a difference in the M3U playlist.

Thanks Matt