volumio/Volumio2 #1764: Can't stream to Volumio from Linux

ImaCrea, 2019-06-04:
Hi there,
I used to stream from Elementary OS using pulseaudio-dlna and it worked well.

But for some reason I can't understand, it doesn't work anymore. 
Pulseaudio-dlna get stuck on :

`06-04 15:14:07 pulseaudio_dlna.discover                       INFO     SSDPDiscover.search()`

It looks like it doesn't find the Volumio device. I have rebooted devices, disconnected VPN, it doesn't work. On my browser, I can connect to the device on 192.168.0.10

I'm not sure where to look for the bug. I have updated to latest Volumio version, is it coming from there? I have re-installed pulseaudio-dlna on my computer but still same problem.

Please help :)

steve80289, 2020-05-01:
Do you still have this issue?

ImaCrea, 2020-05-04:
Hi, I switched to Moode since then. I find it more robust and faster than Volumio.

Cheers and thanks for asking.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐