volumio/Volumio2 #1768: Albums with the same name

llevet, 2019-06-10:
Hi.

remove patch : 
https://github.com/volumio/Volumio2/commit/73603df2a9106c7b1eac6b3cc86c9b00610d37a9

This patch broke adding Albums with the same name, because there is no filter on Artis, so all album with same name is add in queue.

https://forum.volumio.org/albums-with-the-same-name-t5719.html

Ludo.

llevet, 2019-07-13:
Will be corrected by : 
https://github.com/volumio/Volumio2/pull/1770

llevet.

cbergmiller, 2020-04-19:
I think https://github.com/volumio/Volumio2/commit/423217629854b2b73e45c54242eedef5929d70dd introduced a regression on this issue for files with empty `albumartist` fields. When `album` is the same files are added even when `artist` differs.
Or must albums (that aren't compilitions) always have populated `albumartist` fields in all files?

llevet, 2020-04-19:
Your are right, albumartist must be alway populate.
(https://forum.volumio.org/problem-playing-certain-albums-volumio-599-and-598-t12774.html)
According to MPD (kernel of volumio) : 
albumartist: on multi-artist albums, this is the artist name which shall be used for the whole album. The exact meaning of this tag is not well-defined.
(https://www.musicpd.org/doc/html/protocol.html).

And other thing, MPD said : 
find EXPRESSION : AlbumArtist looks for VALUE in AlbumArtist and falls back to Artist tags if AlbumArtist does not exist. VALUE is what to find.
But the UI of volumio can't.

Ludovic.