volumio/Volumio2 #1771: Fixed volume for Spotify Connect

JohanGroenenboom, 2019-06-12:
Here's the use case: Volumio + Spotify Connect plugin + Roon bridge on RPi 3B+. 
Digital output connected to active speakers. Volume control in the speakers.
Multiple users in same household. Some using Spotify, some using Roon. 
Volumio Mixer Type configured to "None".
Now the issue: Given this configuration, Roon sees the Volumio device as a fixed-volume device, but Spotify does not. Spotify can lower the volume from 0dB, and some users will do this. Next they will increase dac/amp volume using the speaker remote control. 
When switching from Spotify to Roon, volume will be scaringly loud. 
Proposed solution: allow volume fixed at 0dB / block volume control from Spotify. 
(as a reference, Sonos device set to fixed volume behaves this way with Spotify).
If you can give me some hints as to where in the system to solve this, I could give it a go and make a contribution.

JohanGroenenboom, 2019-06-12:
P.S. I'm not clear yet if this is a bug report or a feature request - shouldn't the behaviour with Mixer Type set to None result in fixed volume anyway?

ashthespy, 2019-06-12:
When there is no mixer (aka Mixer `None`) then the any changes from other Spotify clients via Spotify Connect are implemented as a pure software solution, keeping the max threshold the current hardware volume.  
The mixer stuff is handled by [`vollibrespot`](https://github.com/ashthespy/Vollibrespot) and [`librespot`](https://github.com/ashthespy/librespot/tree/vollibrespot), but there isn't an option to disable volume control, but feel free to give it a shot!

I don't have a Sonos device, but how does a Spotify client behave with a fixed volume device? Does the volume bar grey out or something?

JohanGroenenboom, 2019-06-24:
Thanks for your pointers. To answer your question, he Sonos device with fixed output setting ignores all volume commands. Some clients show this by disabling the control in the GUI, but Spotify volume widget still works, it just has no effect on the Sonos. 
I have implemented a fixed-volume setting on librespot which works for me. But I just found https://github.com/librespot-org/librespot/pull/286 which implements a null-mixer which does the same thing. I prefer the null-mixer solution - better application of the Open Closed principle - so I won't submit my changes to librespot.

ashthespy, 2019-07-08:
As mentioned over at PR, have implemented this feature (along with disabling the client UI), and it should make it into the next release of the plugin.

closed 2021-01-06 by ashthespy

freshmas, 2021-09-19:
This is still not solved?

ashthespy, 2021-09-20:
> This is still not solved?

Could you elaborate?

freshmas, 2021-09-20:
When I select Volumio in Spotify, the volume comes up at some low
percentage ~10%. This must be manually increased each time a user connects,
and their device volume is held hostage. If they are browsing on their
phone and a video starts playing, it is at full volume. If they turn their
device volume down with the hardware rocker, the volumio output also
decreases.

ashthespy, 2021-09-20:
How the Spotifiy mobile app works can't really be controlled by Volumio.. If you want to avoid this, just use a fixed predefined volume as this issue details.

volumio, 2021-09-20:
When starting, IMHO spotify connect shall ask Volumio the current volume level (via REST API) then set its internal volume at this value... Do you think it's achievable?

freshmas, 2021-09-20:
The extra details in my post may have confused the central fact that the output of Volumio is not fixed at 100% as it should be.

freshmas, 2021-09-20:
> When starting, IMHO spotify connect shall ask Volumio the current volume level (via REST API) then set its internal volume at this value... Do you think it's achievable?

That sounds like a helpful feature for when the Volumio mixer is enabled. When set to ‘none’ it shouldn’t matter what the spotify volume is set to. Volumio should provide 100% output to the user’s preamp.

ashthespy, 2021-09-20:
> The extra details in my post may have confused the central fact that the output of Volumio is not fixed at 100% as it should be.

This is already the case when Mixer is set to `None` in Volumio..

ashthespy, 2021-09-20:
> When starting, IMHO spotify connect shall ask Volumio the current volume level (via REST API) then set its internal volume at this value... Do you think it's achievable?

IIRC, the daemon volume would be the last volume of the Volumio device (if the initial volume setting is not enabled)

freshmas, 2021-09-20:
I just did a quick test.

Booted Volumio: volume level greyed out and 100%

Connected through Spotify: volume level set to 20%
Cannot change in Volumio UI.
Changes made through Spotify or with iPhone volume rocker apply to Volumio UI and volume output.
Changes made in Volumio UI briefly appear but then snap back to previous value.

ashthespy, 2021-09-20:
What is your volume control set to in the plug-in configurration page? Set it to `fixed`..

freshmas, 2021-09-20:
Respect. That was the problem. There was also an update to the plug-in.
Thank you for your help! I remember setting that up a couple years ago but I hadn’t used it for a while.

ashthespy, 2021-09-20:
>  Thank you for your help! I remember setting that up a couple years ago but I hadn’t used it for a while.

The curse of flexibility - more settings to play with!  We try to provide some sane defaults, but everyone have varying definitions of sane ;-)

freshmas, 2021-09-20:
This has been much better than raspotify was. It kept breaking every few weeks.
This one my way fault for not remembering the last time I used this streamer was with a power amp.