volumio/Volumio2 #1774: Volumio doesn't start after update to latest version

CherryDT, 2019-06-17:
```
Jun 17 06:22:13 volumio volumio[2261]: info: Loading plugins from folder /data/myvolumio/plugins
Jun 17 06:22:13 volumio volumio[2261]: info: Loading plugin "my_volumio"...
Jun 17 06:22:13 volumio volumio[2261]: info: Loading plugin "cd_controller"...
Jun 17 06:22:13 volumio volumio[2261]: info: Loading plugin "streaming_services"...
Jun 17 06:22:13 volumio volumio[2261]: info: CoreCommandRouter::executeOnPlugin: system , getConfigParam
Jun 17 06:22:13 volumio volumio[2261]: info: CoreCommandRouter::executeOnPlugin: system , getConfigParam
Jun 17 06:22:13 volumio volumio[2261]: info: Starting MyVolumio Remote Streaming Endpoints
Jun 17 06:22:13 volumio volumio[2261]: info: Adding CD REST API Endpoints
Jun 17 06:22:13 volumio volumio[2261]: /myvolumio/plugins/music_service/cd_controller/cd_controller_real:1
Jun 17 06:22:13 volumio volumio[2261]: (function (exports, require, module, __filename, __dirname) { var exec=require("child_process").exec,execSync=(exec=require("child_process").exec,require("child_process").execSync),fs=require("fs-extra"),libQ=require("kew"),os=require("os"),mediaDir="",cdAlbum="",cdArtist="",cdSpeed="24",isRipping=!1,abcdeDebug=!1,cddbFolder="/data/cddb/",variant="volumio";function ControllerCD(e){this.context=e,this.config=new(require("v-conf")),this.logger=this.context.logger,this.commandRouter=this.context.coreCommand}module.exports=ControllerCD,ControllerCD.prototype.onVolumioStart=function(){var e=libQ.defer(),t=this.commandRouter.pluginManager.getConfigurationFile(this.context,"config.json");return this.config.loadFile(t),e.promise},ControllerCD.prototype.onStart=function(){var e=libQ.defer();this.commandRouter.sharedVars.get("language_code");return this.commandRouter.sharedVars.registerCallback("language_code",this.systemLanguageChanged.bind(this)),t
Jun 17 06:22:13 volumio volumio[2261]: TypeError: this.commandRouter.addPluginRestEndpoint is not a function
Jun 17 06:22:13 volumio volumio[2261]: at ControllerCD.addCDRestEndpoints (/myvolumio/plugins/music_service/cd_controller/cd_controller_real:1:13986)
Jun 17 06:22:13 volumio volumio[2261]: at ControllerCD.onStart (/myvolumio/plugins/music_service/cd_controller/cd_controller_real:1:1309)
Jun 17 06:22:13 volumio volumio[2261]: at PluginManager.startMyVolumioPlugin (/volumio/app/pluginmanager.js:395:36)
Jun 17 06:22:13 volumio volumio[2261]: at HashMap.<anonymous> (/volumio/app/pluginmanager.js:426:35)
Jun 17 06:22:13 volumio volumio[2261]: at HashMap.forEach (/volumio/node_modules/hashmap/hashmap.js:172:10)
Jun 17 06:22:13 volumio volumio[2261]: at HashMap.proto.(anonymous function) [as forEach] (/volumio/node_modules/hashmap/hashmap.js:184:7)
Jun 17 06:22:13 volumio volumio[2261]: at PluginManager.startMyVolumioPlugins (/volumio/app/pluginmanager.js:425:27)
Jun 17 06:22:13 volumio volumio[2261]: at PluginManager.startPlugins (/volumio/app/pluginmanager.js:83:10)
Jun 17 06:22:13 volumio volumio[2261]: at new CoreCommandRouter (/volumio/app/index.js:61:24)
Jun 17 06:22:13 volumio volumio[2261]: at Object.<anonymous> (/volumio/index.js:35:21)
Jun 17 06:22:13 volumio volumio[2261]: at Module._compile (module.js:652:30)
Jun 17 06:22:13 volumio volumio[2261]: at Object.Module._extensions..js (module.js:663:10)
Jun 17 06:22:13 volumio volumio[2261]: at Module.load (module.js:565:32)
Jun 17 06:22:13 volumio volumio[2261]: at tryModuleLoad (module.js:505:12)
Jun 17 06:22:13 volumio volumio[2261]: at Function.Module._load (module.js:497:3)
Jun 17 06:22:13 volumio volumio[2261]: at Function.Module.runMain (module.js:693:10)
Jun 17 06:22:13 volumio systemd[1]: volumio.service: main process exited, code=exited, status=1/FAILURE
Jun 17 06:22:13 volumio systemd[1]: Unit volumio.service entered failed state.
```

CherryDT, 2019-06-17:
Workaround: Move folder ` /myvolumio/plugins/music_service/cd_controller` elsewhere, e.g. `/home/volumio/bad_plugin`

volumio, 2019-09-11:
Your install seems corrupted or you manually edited files

closed 2019-09-11 by volumio