volumio/Volumio2 #1775: /volumio/.globals.json ENOENT - Volumio doesn't start after update to latest version 2.587

CherryDT, 2019-06-17:
This is not the same as #1774 - this is a different error on another volumio machine!

```Jun 17 07:16:48 volumio volumio[1238]: /volumio/node_modules/fs-extra/node_modules/jsonfile/index.js:73
Jun 17 07:16:48 volumio volumio[1238]: throw err
Jun 17 07:16:48 volumio volumio[1238]: ^
Jun 17 07:16:48 volumio volumio[1238]: Error: /volumio/.globals.json: ENOENT: no such file or directory, open '/volumio/.globals.json'
Jun 17 07:16:48 volumio volumio[1238]: at Object.fs.openSync (fs.js:646:18)
Jun 17 07:16:48 volumio volumio[1238]: at Object.fs.readFileSync (fs.js:551:33)
Jun 17 07:16:48 volumio volumio[1238]: at Object.readFileSync (/volumio/node_modules/fs-extra/node_modules/jsonfile/index.js:67:22)
Jun 17 07:16:48 volumio volumio[1238]: at Object.<anonymous> (/volumio/index.js:3:18)
Jun 17 07:16:48 volumio volumio[1238]: at Module._compile (module.js:652:30)
Jun 17 07:16:48 volumio volumio[1238]: at Object.Module._extensions..js (module.js:663:10)
Jun 17 07:16:48 volumio volumio[1238]: at Module.load (module.js:565:32)
Jun 17 07:16:48 volumio volumio[1238]: at tryModuleLoad (module.js:505:12)
Jun 17 07:16:48 volumio volumio[1238]: at Function.Module._load (module.js:497:3)
Jun 17 07:16:48 volumio volumio[1238]: at Function.Module.runMain (module.js:693:10)
Jun 17 07:16:48 volumio systemd[1]: volumio.service: main process exited, code=exited, status=1/FAILURE
Jun 17 07:16:48 volumio systemd[1]: Unit volumio.service entered failed state.```

CherryDT, 2019-06-17:
Workaround: Create `/volumio/.globals.json` with contents `{"exitOnException":true}`

volumio, 2019-09-11:
Your install seems corrupted or you manually edited files

closed 2019-09-11 by volumio