volumio/Volumio2 #1779: Need proper MPD integration

YohannLYD, 2019-07-13:
Hi,

I mostly use MPD clients to control Volumio but I noticed that when I noticed that when I play a track it isn't updated on the Volumio webui. This isn't such a big issue for me.

But when a continuous stream such as a webradio is playing and I try to stop or replace it by another stream/file the current stream just stops and restarts again.

Has there been a consideration to improve the use of Volumio through MPD clients ?

h4de5, 2019-10-06:
I think I have the same problem here. the playlist progression between what you can see in volumio and what is played from mpc differs.
maybe the volumio command line needs a way to manage items. 
(maybe we will be able to add upnp sources via command line as well?)

jwoodard80, 2020-03-18:
I have this same issue, I'm using home assistant to send mpd files for TTS. Webradio pauses, does not play the file, and then resumes.

Currently I have to have 2 Pi's on my network just to send TTS to the dedicated one. I would be happy to help troubleshoot this if someone could suggest where to start.

v-evets, 2023-06-30:
And to think was actually considering a premium sub. 

Questionable architecture decision to emulate MPD on top of MPD, supposedly to support "streaming" services I don't care about (and other SBC MPD distros somehow manage just fine without such borkage)... Yet said emulation remains half-implemented and pretty much useless since **2016** #494 and still just as broken in volumio 3.

I mean, I'd probably have been willing to help where I could with something like this, but between the apparent total lack of interest in this basic functionality and the premium nags and paywalls in the new versions... Yeah, nah.

Shame that, guess I'll go install rune or moode, or maybe just mopidy on Debian.

volumio, 2023-06-30:
This feature was never implemented because there was near zero demand. All of the rest you say are imho very questionable opinions