volumio/Volumio2 #1780: No Picutres in Artist view

baensch, 2019-07-14:
Since last month my Volumio dont shows me the Artist Picture. It is only a Gray Star on a White Background. I attached a Picture. 
[image]

himynameisfuiz, 2019-08-03:
I got the same problem for a few weeks now... seems like it has to do with last.fm (the source of the artist images) changing their API. Was also discussed on the official Volumio board: https://forum.volumio.org/artist-art-t12359.html

_"We’ve made changes to our API in order to limit abuse and improve the service for everyone, in line with our API Terms of Use. While we allow API users to have access to a lot of data through the API, it has always been against our API Terms of Use for third parties to use audio, audiovisual, images or artwork. In some instances, some data may have been indirectly accessible through a small number of API calls, and so as part of the recent API clean-up, we have corrected that anomaly."_

Anyway, would be great to get a fix ASAP as it really takes a lot of the great look and feel Volumio usually has to offer.

himynameisfuiz, 2019-08-21:
I still really hope this gets fixed ASAP as it really feels weird to sort by artist without having different artist images.

volumio, 2019-09-11:
Fixed

closed 2019-09-11 by volumio