volumio/Volumio2 #1783: Noise occurs in audio when volume is greater than 85%

gvijayakumar-16, 2019-07-17:
Hi,

There is background noise in audio whenever the volume is increased above 85%

I have volumio running on raspberry pi 3 b+

dan90210, 2019-08-02:
Are you using the built in headphone jack? The quality of the jack on the pi is honestly terrible and you will most likely experience distortion with anything above 85%. I solved this problem by connecting to an external dac via USB.

gvijayakumar-16, 2019-08-03:
Can't say as I don't have a USB dac handy at the moment. And yes, I am using the headphone jack.

himynameisfuiz, 2019-08-04:
Had this problem when I connected my Pi 3 with my Amazon Echo Plus line in port, even though I use a HifiBerry DAC+, so it seems as if the problem could also be the receiver. I would recommend to check if other devices also get that kind of background noise when you connect your Pi to them. Doesn't necessarily have to do with using the built in headphone jack in my opinion.

steve80289, 2020-05-01:
I found that if you raised the volume on the speaker you are using (or whatever) and then just lowered the volume on the app side, then everything works better.  I am using the default phone jack.  Before I tried this, if I went above 70%, the sound would crackle.  Now I do not hear it anymore (nor do I go that high for fear of going deaf).  

Hope this helps,

Steve

Wdavery, 2020-06-02:
My Pi3 B+ has exactly this issue. Definitely a hardware issue, it's awful. I set max volume to 80% as a temporary solution before deciding what hardware to use going forward. Probably safe to close this issue.

gvijayakumar-16, 2020-06-03:
Closing this issue as it does not occur once I connected my dac

closed 2020-06-03 by gvijayakumar-16