volumio/Volumio2 #1788: Error message: The selected output device is not available

gvolt, 2019-08-05:
Every now and there is the error message

"The selected output device is not available
Please connect HifiBerry DAC Plus to the unit or select a different output device from playback options"

as reported [here](https://forum.volumio.org/please-access-version-587-2019-t12696.html#p65261).

Sound is working nonetheless. My DAC is an original Hifiberry DAC Plus. Volumio version is 2.599 (fresh install). The error message seems to appear after booting, but as it happens unpredictably I have not been able to capture a log so far.

oisteink, 2019-11-21:
I'm getting this as well - like after every few songs. Started today.

Setting: Playing from nas over nfs, also happens when i play from tidal.
Only have 1 plugin: last.fm scrobbler
Connected DAC is SupTronics X5000 configured according to http://www.suptronics.com/miniPCkits/x5000-software.html

Logs attached - I started playing Ane Brun's album "Spending Time with Morgan" at 20:20 in these logs. Issue happened after first song. If I press play after one song it starts normally.

[mpd.log](https://github.com/volumio/Volumio2/files/3876645/mpd.log)
[volumio.log](https://github.com/volumio/Volumio2/files/3876646/volumio.log)

oisteink, 2019-11-23:
Seems like it was the last.fm scrobble plugin that was messing it up for me. I stopped using it and started using mpdscribble instead and have no issues now