volumio/Volumio2 #1792: MPD traceback on "No such directory"

arthurlutz, 2019-08-24:
```
2019-08-24T17:06:03.415Z - info:  message= [50@0] {} No such directory, stack=Error:  [50@0] {} No such directory
    at MpdClient.receive (/volumio/app/plugins/music_service/mpd/lib/mpd.js:63:14)
    at Socket.<anonymous> (/volumio/app/plugins/music_service/mpd/lib/mpd.js:43:10)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at addChunk (_stream_readable.js:263:12)
    at readableAddChunk (_stream_readable.js:246:13)
    at Socket.Readable.push (_stream_readable.js:208:10)
    at Pipe.onread (net.js:607:20)
```

volumio, 2019-09-11:
Can you provide a context? when does this happen?

oisteink, 2019-11-24:
Happens during startup for me - log here: http://logs.volumio.org/volumio/aeeLY2Z.html

```
Nov 24 03:39:17 volumio volumio[821]: info: Reloading queue from file
Nov 24 03:39:17 volumio volumio[821]: info:
Nov 24 03:39:17 volumio volumio[821]: info:  message= [50@0] {} No such directory, stack=Error:  [50@0] {} No such directory
Nov 24 03:39:17 volumio volumio[821]: at MpdClient.receive (/volumio/app/plugins/music_service/mpd/lib/mpd.js:63:14)
Nov 24 03:39:17 volumio volumio[821]: at Socket. (/volumio/app/plugins/music_service/mpd/lib/mpd.js:43:10)
Nov 24 03:39:17 volumio volumio[821]: at emitOne (events.js:116:13)
Nov 24 03:39:17 volumio volumio[821]: at Socket.emit (events.js:211:7)
Nov 24 03:39:17 volumio volumio[821]: at addChunk (_stream_readable.js:263:12)
Nov 24 03:39:17 volumio volumio[821]: at readableAddChunk (_stream_readable.js:246:13)
Nov 24 03:39:17 volumio volumio[821]: at Socket.Readable.push (_stream_readable.js:208:10)
Nov 24 03:39:17 volumio volumio[821]: at Pipe.onread (net.js:607:20)
Nov 24 03:39:17 volumio volumio[821]: info: CoreStateMachine::setRepeat null single undefined

```

volumio, 2019-11-27:
Not an issue, it just checks for USB

closed 2019-11-27 by volumio