volumio/Volumio2 #1793: Can't add files from mounted usb volume

xamanu, 2019-09-01:
Thank you for this great piece of software.

When adding an external hard drive formatted with ext4 over user, it get's successfully mounted under `/media/music` and when sshing into the Raspberry Pi, I can see the files there.
But I'm not able to add it to Volumio. Rescanning under `http://volumio.local/plugin/miscellanea-my_music` won't bring it up.

Can you help me to debug this, please? Thanks.

JayH1119, 2019-09-02:
I might be experiencing the same problem. Raspberry Pi 4, Model B; Volumio 2.599. Adding an external 4TB drive with an extensive library in FLAC, the library apparently updates as the circular icon rotates  on the browse tab and when it finishes it shows: 575 Artists, 3327 Albums, 59975 Tracks and 7277:14:22 play time. There are no errors in mpd.log but when I go to Browse tab, there is nothing shown/available.

JayH1119, 2019-09-02:
never mind, the browse list populated overnight. I was expecting some type of indicator that the system was still working on something.

xipmix, 2019-09-27:
Yes that would be nice. See also volumio/Volumio2-UI#513