volumio/Volumio2 #1803: Pop-up info box lags when deleting from queue

bjornarsu, 2019-09-18:
An annoying issue: when removing songs from the queue, there's a green floating info box that pops up at the top of the UI. This box has a countdown progress bar, and stays there until the progress bar times out. 

When deleting many songs from the queue quickly, the green box pops up and waits for a fixed timeout for each single delete action. 

This builds up a long queue of these pop-up boxes that lag a lot since it's waiting for each one to complete. 

They even still keep on popping up when I already have moved to another volumio page.

volumio, 2019-09-19:
You're right, we should show only one toast . Going to work on it

volumio, 2019-09-19:
Fixed with https://github.com/volumio/Volumio2-UI/commit/82fb2065f53189500860cf6eb2737da8567e6ddc
thanks for reporting

closed 2019-09-19 by volumio

bjornarsu, 2019-09-19:
Thanks for fixing!

sandronoel, 2021-08-19:
Can we have a way to disable that box altogether? - it is redundant information.