volumio/Volumio2 #1806: interrupt and reset music scan from the NAS

giuseppedigiulio, 2019-09-20:
I have just purchased a Pro-ject Stream box S2 ultra with the customized software version 1.038. The device recognizes the NAS correctly mounts the music folder and starts scanning the tracks in the folder. When it comes close to 1000 songs it resets the counter and doesn't build the catalog. What's the problem?

volumio, 2019-09-20:
Hi, do this.
Start the scan, then when it stops, send the logs (see: https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html )
and paste the link to the log and send via mail to info@volumio.org I will help from there

giuseppedigiulio, 2019-09-21:
Hi, I send you the link via email.

volumio, 2019-09-23:
Following up via mail

closed 2019-09-23 by volumio