volumio/Volumio2 #1808: Websocket replaceAndPlay fails when index is 0

gilphilbert, 2019-09-23:
When you pass the websocket interface replaceAndPlay using the list and index pragma. The call fails and the queue is left empty. Passing any other index succeeds.

`io.emit('replaceAndPlay',{list: [{…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}], index: 0})
`

I watched the logs (tail -f /var/log/volumio.log) and it seems that this request actually crashes the volumio service:
```
2019-09-23T20:44:36.528Z - info: CoreCommandRouter::volumioReplaceandPlayItems
2019-09-23T20:44:37.604Z - info: -------------------------------------------
2019-09-23T20:44:37.605Z - info: -----            Volumio2              ----
2019-09-23T20:44:37.605Z - info: -------------------------------------------
2019-09-23T20:44:37.606Z - info: -----          System startup          ----
2019-09-23T20:44:37.606Z - info: -------------------------------------------
2019-09-23T20:44:37.617Z - info: MYVOLUMIO Environment detected
2019-09-23T20:44:37.621Z - info: Plugin folders cleanup
2019-09-23T20:44:37.621Z - info: Scanning into folder /volumio/app/plugins/
```
(rest of boot process)

I know Javascript but I don't know much about debugging node environments, so I can't work out how to tell where the bug is.

xipmix, 2019-09-27:
You can get a long way with extra logging statements.
Looking at ```volumioReplaceandPlayItems``` it seems you should  hit this case:
```
    } else if (data.list && data.index) {                                       
        this.stateMachine.addQueueItems(data.list)                              
            .then(()=>{                                                         
                this.volumioPlay(data.index);                                   
                defer.resolve();                                                
            });                                                                 
```
It looks like the call to ```volumioPlay(data.item)``` doesn't happen, but a few more (temporary)```this.pushConsoleMessage``` could make the flow clearer. Also it can be useful to dump out the data structure to make sure it got to the function unscathed.

xipmix, 2019-09-27:
Hm, yes you should be seeing this log line
```
CoreStateMachine::play index 0
```
from ```CoreStateMachine.prototype.play```.

gilphilbert, 2019-09-27:
Yes, what's odd is I tried putting in some extra pushConsoleMessage lines and none of them printed. I've misplaced my dev machine (long story) but should have it back Monday night. I'll add more troubleshooting lines and see what I can find out about the data format.

volumio, 2019-09-30:
@gilphilbert index starts from 1

gilphilbert, 2019-10-01:
> @gilphilbert index starts from 1
Is this also true for replaceAndPlay? If I set the index to 1, it plays the second track. Setting the index to 0 crashes Volumio.

Got my dev machine back, so I inserted a logging statement as follows:
```
                        connWebSocket.on('replaceAndPlay', function (data) {
self.logger.info(JSON.stringify(data));
                                return self.commandRouter.replaceAndPlay(data)
                        })
```
(I had to use self.logger because this.pushConsoleMessage didn't output anything)
This is the result when the index is set to zero (with a crash)

`2019-10-01T03:47:32.553Z - info: {"list":[{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/01. Let It Rock - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/02. You Give Love A Bad Name - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/03. Livin' On A Prayer - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/04. Social Disease - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/05. Wanted Dead Or Alive - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/06. Raise Your Hands - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/07. Without Love - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/08. I'd Die For You - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/09. Never Say Goodbye - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/10. Wild In The Streets - Bon Jovi.flac"}],"index":0}`

The following output is from a successful playback where the index is set to 1:
`2019-10-01T03:51:11.364Z - info: {"list":[{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/01. Let It Rock - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/02. You Give Love A Bad Name - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/03. Livin' On A Prayer - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/04. Social Disease - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/05. Wanted Dead Or Alive - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/06. Raise Your Hands - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/07. Without Love - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/08. I'd Die For You - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/09. Never Say Goodbye - Bon Jovi.flac"},{"uri":"music-library/NAS/musichd/Bon Jovi/Slippery When Wet/10. Wild In The Streets - Bon Jovi.flac"}],"index":1}`
The song it plays is "You Give Love A Bad Name", which is the _second_ track in the input. I've repeated this with different albums and the effect is the same.

xipmix, 2019-10-02:
Interesting. If I am following the code correctly the ```this.volumioPlay(data.index);``` call ends up up in ```CoreStateMachine.prototype.play = function (index) {``` in statemachine.js, which uses 0-based indexing for the ```currentPosition```; e.g. https://github.com/volumio/Volumio2/blob/ca19027e7213a851102bdfe45312c32c270313ff/app/statemachine.js#L1078

There is a check for out-of-range index in the case where ```index``` is _not_ defined, but if it is defined, ```index``` is used unconditionally- 
https://github.com/volumio/Volumio2/blob/ca19027e7213a851102bdfe45312c32c270313ff/app/statemachine.js#L1086-L1094

Still, we should see a ```CoreStateMachine::play index 0``` log line before that happens, and we don't, which is puzzling. Possibly hack in a pushConsoleMessage just before line 1091 ?

gilphilbert, 2019-10-27:
OK, I finally got around to this - sorry about the long delay.

I'm now even more confused. I added some lines thoughout app/index.js to be a little more verbose. You can see the output here when the index is set to 1 (my logging statements begin with HACK). 
```
2019-10-27T06:07:34.564Z - info: CoreCommandRouter::volumioReplaceandPlayItems
2019-10-27T06:07:34.564Z - info: HACK::Clearing queue
2019-10-27T06:07:34.565Z - info: CoreStateMachine::ClearQueue
2019-10-27T06:07:34.565Z - info: CoreStateMachine::stop
2019-10-27T06:07:34.565Z - info: CoreStateMachine::setConsumeUpdateService undefined
2019-10-27T06:07:34.566Z - info: CorePlayQueue::clearPlayQueue
2019-10-27T06:07:34.566Z - info: CorePlayQueue::saveQueue
2019-10-27T06:07:34.567Z - info: CoreStateMachine::pushEmptyState
2019-10-27T06:07:34.568Z - info: CoreCommandRouter::volumioPushState
2019-10-27T06:07:34.568Z - info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
2019-10-27T06:07:34.592Z - info: CoreCommandRouter::volumioPushQueue
2019-10-27T06:07:34.593Z - info: HACK::Cleared queue
2019-10-27T06:07:34.594Z - info: HACK::Data is:{"list":[{"uri":"music-library/NAS/musichd/Del Amitri/Change Everything/10. As Soon As The Tide Comes In - Del Amitri.flac"},{"uri":"music-library/NAS/musichd/Del Amitri/Change Everything/10. As Soon As The Tide Comes In - Del Amitri.flac"}],"index":1}
2019-10-27T06:07:34.594Z - info: HACK::data={"list":[{"uri":"music-library/NAS/musichd/Del Amitri/Change Everything/10. As Soon As The Tide Comes In - Del Amitri.flac"},{"uri":"music-library/NAS/musichd/Del Amitri/Change Everything/10. As Soon As The Tide Comes In - Del Amitri.flac"}],"index":1}
2019-10-27T06:07:34.595Z - info: HACK::Adding items to queue
2019-10-27T06:07:34.596Z - info: CoreStateMachine::addQueueItems
2019-10-27T06:07:34.597Z - info: CorePlayQueue::addQueueItems
2019-10-27T06:07:34.598Z - info: Adding Item to queue: music-library/NAS/musichd/Del Amitri/Change Everything/10. As Soon As The Tide Comes In - Del Amitri.flac
2019-10-27T06:07:34.599Z - info: Exploding uri music-library/NAS/musichd/Del Amitri/Change Everything/10. As Soon As The Tide Comes In - Del Amitri.flac in service mpd
2019-10-27T06:07:34.650Z - info: Adding Item to queue: music-library/NAS/musichd/Del Amitri/Change Everything/10. As Soon As The Tide Comes In - Del Amitri.flac
2019-10-27T06:07:34.651Z - info: Exploding uri music-library/NAS/musichd/Del Amitri/Change Everything/10. As Soon As The Tide Comes In - Del Amitri.flac in service mpd
2019-10-27T06:07:34.652Z - info: HACK::Added queue items
2019-10-27T06:07:34.665Z - info: ALBUMART /albumart?cacheid=983&web=Del%20Amitri/Change%20Everything/extralarge&path=%2Fmnt%2FNAS%2Fmusichd%2FDel%20Amitri%2FChange%20Everything%2F10.%20As%20Soon%20As%20The%20Tide%20Comes%20In%20-%20Del%20Amitri.flac&metadata=false
2019-10-27T06:07:34.665Z - info: URI /mnt/NAS/musichd/Del Amitri/Change Everything/10. As Soon As The Tide Comes In - Del Amitri.flac
2019-10-27T06:07:34.667Z - info: ALBUMART /albumart?cacheid=983&web=Del%20Amitri/Change%20Everything/extralarge&path=%2Fmnt%2FNAS%2Fmusichd%2FDel%20Amitri%2FChange%20Everything%2F10.%20As%20Soon%20As%20The%20Tide%20Comes%20In%20-%20Del%20Amitri.flac&metadata=false
2019-10-27T06:07:34.668Z - info: URI /mnt/NAS/musichd/Del Amitri/Change Everything/10. As Soon As The Tide Comes In - Del Amitri.flac
2019-10-27T06:07:34.672Z - info: CorePlayQueue::saveQueue
2019-10-27T06:07:34.672Z - info: CoreCommandRouter::volumioPushQueue
2019-10-27T06:07:34.676Z - info: CoreStateMachine::updateTrackBlock
2019-10-27T06:07:34.681Z - info: CorePlayQueue::getTrackBlock
2019-10-27T06:07:34.683Z - info: HACK::Index is 1
2019-10-27T06:07:34.683Z - info: CoreCommandRouter::volumioPlay
2019-10-27T06:07:34.685Z - info: CoreStateMachine::play index 1
2019-10-27T06:07:34.685Z - info: CoreStateMachine::setConsumeUpdateService undefined
2019-10-27T06:07:34.687Z - info: CoreStateMachine::stop
2019-10-27T06:07:34.687Z - info: CoreStateMachine::setConsumeUpdateService undefined
2019-10-27T06:07:36.080Z - info: CoreCommandRouter::executeOnPlugin: system , getSystemVersion
2019-10-27T06:07:36.082Z - info: CoreCommandRouter::executeOnPlugin: system , getConfigParam
```

When the index is set to 0, the only output is "Clearing queue"
```
2019-10-27T06:08:30.643Z - info: CoreCommandRouter::volumioReplaceandPlayItems
2019-10-27T06:08:30.643Z - info: HACK::Clearing queue
2019-10-27T06:08:33.389Z - info: -------------------------------------------
2019-10-27T06:08:33.393Z - info: -----            Volumio2              ----
2019-10-27T06:08:33.393Z - info: -------------------------------------------
2019-10-27T06:08:33.393Z - info: -----          System startup          ----
2019-10-27T06:08:33.394Z - info: -------------------------------------------
```

This seems to mean it's crashing while making the call to clear the queue since the log doesn't even include "CoreStateMachine::ClearQueue" which is just before the statemachine actually clears the queue... but that doesn't make sense as the index doesn't matter at this point.

The only thing I can think of is that this code is being run, through to 1092 in statemachine.js, but the application crashes before the logs are written to /var/log/volumio ...?

Any logging statements I put into statemachine.js aren't output at all during a crash.

xipmix, 2019-10-27:
Can you try printing the data input to ```CoreCommandRouter.prototype.replaceAndPlay```, just before it calls ```statemachine.clearQueue``` ? Possibly you could also short-circuit the exploding code by adding a ```defer.reject()``` like the one in the last ```else``` clause, before the clearQueue. Just while exploring what is happening, of course.

gilphilbert, 2019-10-28:
Found it at last! It's obvious when you see it. The `if` statement is checking to see if the index has been defined using `if (data.list && data.index)`. Of course, when data.index is zero, this is evaluated as false and the wrong code runs resulting in the crash.

I've submitted #1826 to fix this.

PR #1827 (2019-11-18): Replace and play fix
Fixes #1808

Current code checks for the existence of data.index using:
```
} else if (data.list && data.index) {
```
Change to check for data.list!=undefined to avoid the situation where index.list=0 and the test fails when it should pass.
closed 2019-10-28 by gilphilbert