volumio/Volumio2 #1832: [2.657] Volume stuck at 100%, unable to to audio settings.

Akuba-, 2019-10-30:
After updating Volumio using the build in updater I have the problem that my audio settings where reset and I am unable to change them anymore. Volume will always be 100% and after I changed my settings it will not play anymore giving me an ALSA error with device hw0;0 not found.

While reading the log file I noticed this: `Cannot open shared library libasound_module_conf_pulse.so` and I guess this might be the cause of it.

Running Volumio on an Pi2.

Log: http://logs.volumio.org/volumio/q3cpGAY.html

jjjeroenski, 2019-12-25:
Same issue like @Akuba -, even with latest release 2.692
There is a fix if you click save submit on output devices after each reboot.

Using Rpi3b with Justboom Amp HAT
On previous releases never a issue

Log: http://logs.volumio.org/volumio/01XG3E2.html

xipmix, 2019-12-26:
When you say the volume is stuck at 100%, there are a few ways to interpret this.
 * Do you mean the volume-control-in-the-web-interface goes to 100% and you can no longer change it with the web interface controls?
 * Or, do you mean the sound output is maxed out and even though you can change the volume setting in the web interface, this makes no difference to the sound output level ?

@jjjeroenski - how does "save submit on output devices after each reboot" help improve things?

Separate question - does the problem happen with all music sources - music library, webradio, spotify ?

jjjeroenski, 2019-12-26:
@xipmix , i did a reboot just now and this is what i could see:
- notification sound when volume started doesn't work
- volume is at 0% even if in output options, there is startup volume 25.
- i can change manual volume from 0 tot 25 or 30 or ...
- i can start a song on usb storage or webradio, there is also sound, but i can't control sound from now. it' stuck. It's not the volume like you have changed before.

- if i open output options, i have output hardware : i2S dac on and justboom amp boards , i click "save" (notification player restarting) without changing something. --> volume control works again.
BUT it's more quiet than before, it's not the normal volume, but the volume control in the web gui up&down works again.

- you have to do this after each reboot and this is only since one of the latest versions.

i tested with usb storage, webradio and spotify, it's always the same after reboot and it's both less volume than before, for the same volume number.

Volume options:
-Mixer type: 'Hardware'
-Name mixer: 'digital'
Same like multiple posts in the volumio forum : https://forum.volumio.org/after-update-657-volume-control-t13107.html

xipmix, 2019-12-26:
Thanks for all that, it should help find the issue.

jjjeroenski, 2019-12-28:
i did another test with a new sd card and fresh install with latest stable 2.692 without any plugins or config changes, only the amp hat. Had directly the same problem like before with the updates.

Some of the moderators/founders "michelangelo" already replicated the issue in october on 
https://forum.volumio.org/after-update-657-volume-control-t13107-10.html#p68209

It looks like this issue isn't there since latest version but since +/- October releases.

xipmix, 2019-12-28:
Thanks, I noticed volumio mentioned something about replicating it but can't see any relevant commits from around that time.

@jjjeroenski - there is something strange about your log. It should show one of the messages from this code - https://github.com/volumio/Volumio2/blob/27e34cd33f46bedb559f032b49a8ebb85e047f9b/app/plugins/audio_interface/alsa_controller/index.js#L46-L51
but I don't see either. It does appear in Akuba's log, with the error about libasound_module_conf_pulse.so.

This suggests your volumeval is set to 'disabled', but then we don't see the log message a little later in the function - "Creating shared var alsa.outputdevice", so maybe you are not executing the 'onVolumioStart()' function at all.

If you feel like trying some code tweaking, it would be useful to add something like
```
	var configFile = this.commandRouter.pluginManager.getConfigurationFile(this.context, 'config.json');
+     self.logger.info("onVolumioStartup - config file is " + configfile);
	this.config = new (require('v-conf'))();
```
to  app/plugins/audio_interface/alsa_controller/index.js and rebooting.
(make a backup first)

jjjeroenski, 2019-12-28:
Hi, tried this but after change, page doesn't load anymore..

`ControllerAlsa.prototype.onVolumioStart = function () {
	var self = this;

	/*
	edit 28/12/2019, see github: https://github.com/volumio/Volumio2/issues/1832
	var configFile = this.commandRouter.pluginManager.getConfigurationFile(this.context, 'config.json');

	this.config = new (require('v-conf'))();
	*/
	var configFile = this.commandRouter.pluginManager.getConfigurationFile(this.context, 'config.json');
	self.logger.info("onVolumioStartup - config file is " + configfile);
	this.config = new (require('v-conf'))();
	/*end edit 28/12/2019*/
	this.config.loadFile(configFile);

	var volumeval = this.config.get('volumestart');

	var socketURL = 'http://localhost:3000';
	var options = {
		transports: ['websocket'],
		'force new connection': true
	};`

returned back to backup file.
also added new log file:  http://logs.volumio.org/volumio/CYfqwn1.html

xipmix, 2019-12-28:
Thanks for giving it a try. Here is the problem
```
Dec 28 07:51:04 keuken volumio[933]: self.logger.info("onVolumioStartup - config file is " + configfile);
Dec 28 07:51:04 keuken volumio[933]: ^
Dec 28 07:51:04 keuken volumio[933]: ReferenceError: configfile is not defined
```
The variable is ```configFile``` not ```configfile```, sorry about that. If you fix the variable name it should work.

jjjeroenski, 2019-12-29:
Changed variable (modificated renamed file attached) > reboot > could load page.
But again same 'volume' problems, could also use workaround like described above.
Also new logfile: http://logs.volumio.org/volumio/l9oFgTR.html

[index.js.txt](https://github.com/volumio/Volumio2/files/4007882/index.js.txt)

xipmix, 2019-12-29:
Thanks, this is just for debugging. The fix will take longer.

jjjeroenski, 2019-12-29:
ok let me know if you need more debug or testing.

xipmix, 2019-12-31:
The log message is coming from ```ControllerAlsa.prototype.checkMixer()```, which calls ```ControllerAlsa.prototype.getMixerControls``` and that seems to be returning an empty list.

Can you please paste the output from:
```
$ cat /data/configuration/audio_interface/alsa_controller/config.json
```

xipmix, 2019-12-31:
It would also be helpful to see the log with this line added to ```app/plugins/audio_interface/alsa_controller/index.js```
```
ControllerAlsa.prototype.setDefaultMixer  = function (device) {
...
        for (var i in cards) {                                                  
                var devnum = device.toString();                                 
                if ( devnum == cards[i].id) {                                   
                        currentcardname = cards[i].name;                        
                }                                                               
        }                                                                       
+      self.logger.info("card name for device '" + device + "' is '" + currentcardname + "'");
```
The reason is that it's unclear if currentcardname is being assigned any value.

jjjeroenski, 2020-01-02:
> cat /data/configuration/audio_interface/alsa_controller/config.json

volumio@keuken:~$ cat /data/configuration/audio_interface/alsa_controller/config.json
{
  "volumestart": {
    "type": "string",
    "value": "25"
  },
  "volumemax": {
    "type": "string",
    "value": "100"
  },
  "volumecurvemode": {
    "type": "string",
    "value": "logarithmic"
  },
  "outputdevice": {
    "type": "string",
    "value": "1"
  },
  "mixer_type": {
    "type": "string",
    "value": "Hardware"
  },
  "mixer": {
    "type": "string",
    "value": "Digital"
  },
  "volumesteps": {
    "type": "string",
    "value": "5"
  },
  "outputdevicename": {
    "type": "string",
    "value": "JustBoom Amp Boards"
  }
}

tested twice, once after reset without modifications and one time when playing music. Just the same output.

jjjeroenski, 2020-01-02:
> It would also be helpful to see the log with this line added to `app/plugins/audio_interface/alsa_controller/index.js`
> 
> ```
> ControllerAlsa.prototype.setDefaultMixer  = function (device) {
> ...
>         for (var i in cards) {                                                  
>                 var devnum = device.toString();                                 
>                 if ( devnum == cards[i].id) {                                   
>                         currentcardname = cards[i].name;                        
> +      self.logger.info("device " + devnum + " has name '" + currentcardname + "'");
>                 }                                                               
>         }                                                                       
> ```
> 
> The reason is that it's unclear if currentcardname is being assigned the right value.

ok changed, and restarted with same procedure.
new logfile : http://logs.volumio.org/volumio/YR4rc49.html
new index.js(.txt) file attached
[index.js.txt](https://github.com/volumio/Volumio2/files/4017142/index.js.txt)

xipmix, 2020-01-03:
Thanks. Now we step through ```getMixerControls```. What does this command return:
```
$ amixer -c 1 scontents
```
(it would help if you use the 'insert code' icon (<>)when pasting)

While we're there perhaps try to log what gets passed to ```checkValidMixer```:
```
ControllerAlsa.prototype.getMixerControls  = function (device) {
...
+ self.logger.info("mixer set to '" + mixer + "'");
                if (self.checkValidMixer(mixer)) {                              
                    mixers.push(mixer);                                         
                }                                                               
```

jjjeroenski, 2020-01-03:
sorry i forgot to use code icon...
`
volumio@keuken:~$ amixer -c 1 scontents
Simple mixer control 'DSP Program',0
  Capabilities: enum
  Items: 'FIR interpolation with de-emphasis' 'Low latency IIR with de-emphasis' 'High attenuation with de-emphasis' 'Fixed process flow' 'Ringing-less low latency FIR'
  Item0: 'FIR interpolation with de-emphasis'
Simple mixer control 'Analogue',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 1
  Mono:
  Front Left: Playback 0 [0%] [-6.00dB]
  Front Right: Playback 0 [0%] [-6.00dB]
Simple mixer control 'Analogue Playback Boost',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 1
  Front Left: 0 [0%] [0.00dB]
  Front Right: 0 [0%] [0.00dB]
Simple mixer control 'Auto Mute',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [on]
  Front Right: Playback [on]
Simple mixer control 'Auto Mute Mono',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'Auto Mute Time Left',0
  Capabilities: enum
  Items: '21ms' '106ms' '213ms' '533ms' '1.07s' '2.13s' '5.33s' '10.66s'
  Item0: '21ms'
Simple mixer control 'Auto Mute Time Right',0
  Capabilities: enum
  Items: '21ms' '106ms' '213ms' '533ms' '1.07s' '2.13s' '5.33s' '10.66s'
  Item0: '21ms'
Simple mixer control 'Clock Missing Period',0
  Capabilities: enum
  Items: '1s' '2s' '3s' '4s' '5s' '6s' '7s' '8s'
  Item0: '1s'
Simple mixer control 'Deemphasis',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'Digital',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 207
  Mono:
  Front Left: Playback 144 [70%] [-31.50dB] [on]
  Front Right: Playback 144 [70%] [-31.50dB] [on]
Simple mixer control 'Max Overclock DAC',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 40
  Mono: 0 [0%]
Simple mixer control 'Max Overclock DSP',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 40
  Mono: 0 [0%]
Simple mixer control 'Max Overclock PLL',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 20
  Mono: 0 [0%]
Simple mixer control 'Volume Ramp Down Emergency Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Down Emergency Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '4dB/step'
Simple mixer control 'Volume Ramp Down Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Down Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '1dB/step'
Simple mixer control 'Volume Ramp Up Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Up Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '1dB/step
'

`

jjjeroenski, 2020-01-03:
> ControllerAlsa.prototype.getMixerControls  = function (device) {
> ...
> + self.logger.info("mixer set to '" + mixer + "'");
>                 if (self.checkValidMixer(mixer)) [...]

new logfile after adding the line : http://logs.volumio.org/volumio/sxUWCAT.html

xipmix, 2020-01-03:
That's fine, thank you. The log shows we never hit that new log line, which is a surprise.
Time for a bit more thought.

jjjeroenski, 2020-01-03:
ok let me know if you need more

xipmix, 2020-01-03:
Can we check what defaultmixer is set to before it decides whether to try calling amixer?
```
ControllerAlsa.prototype.getMixerControls = function (device) {
...
+ self.logger.info("defaultmixer is  '" + defaultmixer + "'");
    if (defaultmixer) {                                                         
        this.mixertype = 'Hardware';                                            
    } else {                                                                    
```
Thanks

jjjeroenski, 2020-01-03:
added line + reboot + same procedure + new logging : http://logs.volumio.org/volumio/RfYs7eA.html

xipmix, 2020-01-04:
So the amixer output should be getting parsed into an array like this:
```
['Analogue']
['Analogue Playback Boost']
['Auto Mute']
['Auto Mute Mono']
['Deemphasis']
['Digital']
['Max Overclock DAC']
['Max Overclock DSP']
['Max Overclock PLL']
```
but it seems it isn't.

I'll attach a version of the file for you to try, which includes your changes so far and some other logging. You can compare with 'diff -u oldversion.js newversion.js'.

xipmix, 2020-01-04:
[index.js.txt](https://github.com/volumio/Volumio2/files/4021591/index.js.txt)

xipmix, 2020-01-04:
Just for reference when I run the amixer card (no DAC) I get
```
$ amixer -c 0 scontents
Simple mixer control 'PCM',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback -2336 [74%] [-23.36dB] [on]
```
which should parse to
```
['PCM']
```

jjjeroenski, 2020-01-04:
- backup index.js
- replaced index.js in folder /volumio/app/plugins/audio_interface/alsa_controller/
- restart 
- sounds works normal without workaround
- new debug : http://logs.volumio.org/volumio/W9VPzhO.html

jjjeroenski, 2020-01-04:
> Just for reference when I run the amixer card (no DAC) I get
> 
> ```

output amixer:

`$ amixer -c 0 scontents
Simple mixer control 'PCM',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback -2000 [77%] [-20.00dB] [on]
`

xipmix, 2020-01-04:
well - that's good news but a bit surprising. All I did was add debug lines ... and one lonely 
```var self = this;```.
Ah - that's it! Because later in the function we have
```
                if (self.checkValidMixer(mixer)) {                              
                    mixers.push(mixer);                                         
                }                                                               
```
Before, ```self``` was not defined so that condition was never true.

Could you paste again please
```
$ cat /data/configuration/audio_interface/alsa_controller/config.json
```
just to check if something changed there.

I'll send a PR for the fix. Can you test the attached version please - all changes removed except the fix.
[index.js.txt](https://github.com/volumio/Volumio2/files/4022686/index.js.txt)

jjjeroenski, 2020-01-05:
> Could you paste again please
> 
> ```
> $ cat /data/configuration/audio_interface/alsa_controller/config.json
> ```

looks still the same: 

$ cat /data/configuration/audio_interface/alsa_controller/config.json
{
  "volumestart": {
    "type": "string",
    "value": "25"
  },
  "volumemax": {
    "type": "string",
    "value": "100"
  },
  "volumecurvemode": {
    "type": "string",
    "value": "logarithmic"
  },
  "outputdevice": {
    "type": "string",
    "value": "1"
  },
  "mixer_type": {
    "type": "string",
    "value": "Hardware"
  },
  "mixer": {
    "type": "string",
    "value": "Digital"
  },
  "volumesteps": {
    "type": "string",
    "value": "5"
  },
  "outputdevicename": {
    "type": "string",
    "value": "JustBoom Amp Boards"
  }
}

jjjeroenski, 2020-01-05:
> I'll send a PR for the fix. Can you test the attached version please - all changes removed except the fix.
> [index.js.txt](https://github.com/volumio/Volumio2/files/4022686/index.js.txt)

volume control after reboot still works with your fix, thanks! 
for your info, new debug log since last reboot: http://logs.volumio.org/volumio/FxrE9Ve.html

_i don't know if i have to mention here, and even not so important for me but: 
startup notification sound doesn't work with the external amplifier. It is enabled in the settings bud doesn't hear this, it was working before enabling output amp._

xipmix, 2020-01-05:
Good news!
The startup sound is likely a separate problem. If you would like it to be fixed please check for an open issue and if there's none please add an issue. You could paste your latest log url from above in, to get people started.

jjjeroenski, 2020-01-05:
allright @xipmix thanks for your time and efforts!

jjjeroenski, 2020-01-05:
> The startup sound is likely a separate problem. If you would like it to be fixed please check for an open issue and if there's none please add an issue. You could paste your latest log url from above in, to get people started.

#708

closed 2020-01-07 by volumio

damien122, 2020-01-16:
When will be the next volumio release with this fix inside?

volumio, 2020-01-16:
This weekend

cprezzi, 2020-06-03:
I have updated to 2.777 today and get the same problem. Volume can be changed on web-frontend but no change on analog output.

PR #1872 (2020-01-07): Ensure checkValidMixer actually gets called
With 'self' not defined, the self.checkValidMixer condition is always false
and no mixers get added to the mixers array.

Fixes: #1832

Tested on a rpi with no Dac, no explosions.
Testing on an rpi with a Dac should be done soon - done, seems to fix the issue.
commit fd5cf072 (2020-01-07): Ensure checkValidMixer actually gets called

With 'self' not defined, the self.checkValidMixer condition is alwas false
and no mixers get added to the mixers array.

Fixes: #1832