volumio/Volumio2 #1835: Random Restart and Update Issues

jammininmypjs, 2019-11-05:
For some reason, whenever I try to click on the Radiolab podcast (http://feeds.wnyc.org/radiolab) the raspberry pi restarts
Also, whenever I try to update or switch to Volumio3, my HiFiBerry Dac stops working and I have to reflash the sd card
I also noticed favorite radio stations don't persist, however that might be intentional

volumio, 2019-11-06:
Hi. Can you send logs?
https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html

Paste the link on an email to info at volumio dot org

jammininmypjs, 2019-11-06:
http://logs.volumio.org/volumio/pSXGUfH.htmlIf there are other logs or something that would be helpful, let me know!Thanks,Kelly Sent from Mail for Windows 10 From: VolumioSent: Wednesday, November 6, 2019 5:58 AMTo: volumio/Volumio2Cc: jammininmypjs; AuthorSubject: Re: [volumio/Volumio2] Random Restart and Update Issues (#1835) Hi. Can you send logs?https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.htmlPaste the link on an email to info at volumio dot org

volumio, 2019-11-06:
Thanks for sending the log. It appears to be an issue with the podcast plugin. I will contact the developer of the plugin. Will continue here: https://github.com/volumio/volumio-plugins/issues/374

closed 2019-11-06 by volumio

jammininmypjs, 2019-11-06:
> volumio/volumio-plugins#374

Thank you!