volumio/Volumio2 #1836: Pluginmager - Handeling callbacks

ashthespy, 2019-11-06:
For plugins that need access to the configured output device, the current way seems to be via `commandRouter.sharedVars.registerCallback`, as in `airplay_emulation`:
https://github.com/volumio/Volumio2/blob/0e846201a5a7f222da3ed4356fdcbb7742fefc1c/app/plugins/music_service/airplay_emulation/index.js#L47-L48

These callbacks are however not disabled when the plugin is disabled/stopped. Thus causing undesired behaviour. 

I took a look at [`v-conf`](https://github.com/fanciulli/v-conf/blob/master/index.js#L396), but there is no method to cancel the callbacks.  What is the recommended way to handle this?

volumio, 2019-11-07:
You are right... And now that I think, there is no way to cancel those callbacks. We need to find a solution for that

ashthespy, 2019-11-07:
We could always track the plugin enabled/disabled state internally, and check that before reacting to the callbacks, but that would be a fix just for the symptoms...

ashthespy, 2022-01-22:
Repo (but not old issues) has been moved over to https://github.com/volumio/volumio3-backend

closed 2022-01-22 by ashthespy