volumio/Volumio2 #1838: broadcastMessage('pushUiConfig', config) breaks "visibleif" elements

patrickkfkan, 2019-11-10:
Refreshing a UI configuration page through commandRouter.broadcastMessage('pushUiConfig'...) breaks UI elements that have 'visibleif' property. After refresh, the 'visibleif' property is no longer honored.

E.g.:
- In Playback Options, the bit-depth, resampling rate and quality fields will show or hide depending on the on/off state of the Audio Resampling switch.
- Now change mixer type to a different value (say 'hardware' to 'software') and save. Page will refresh automatically to show options relevant to the selected mixer type.
- Now toggle the Audio Resampling switch. The visibility of the bit-depth, resampling rate and quality fields no longer changes along with the state of the switch.

Probably a minor nuisance, but can be confusing to the novice user.

volumio, 2019-11-11:
Thanks for the report. This looks like a UI issue, coul you please close it here and open it in the UI repo?

patrickkfkan, 2019-11-11:
Ahh..yes. My bad.

closed 2019-11-11 by patrickkfkan