volumio/Volumio2 #1844: webradio playback status problem (conflicts with volspotconnect2)

dmnc-net, 2019-11-18:
**Steps to reproduce:**
1. with Spotify Connect (using @balbuze `volspotconnect2`) start playing a song
2. in Volumio-UI overtake the playback by playing some `webradio` stream
3. watching via CLI `volumio status` ... results in Symptoms section
4. at this moment other plugins like thinks there is no playback and trying to play its content but webradio is still playing

**Symptoms:**
CLI command `volumio status`:
`"status"` became `"play"` for a quick moment and then turns to `"stop"` but music is playing and `"seek"` is changing. Also icon in the UI shows ▶ like if there is no playback.

**Background:**
If Volumio is used by multiple users, some of them are using Volumio UI for webradio, other just Spotify Connect thru plugin, this could be a issue.

volumio, 2019-11-18:
Thanks for reporting. This is a volspotconnect issue, rather than volumio's.
Please report into volumio-plugins

@balbuze please see

closed 2019-11-18 by volumio

ashthespy, 2019-11-20:
@dmnc-net  What version of Volumio are you on? This shouldn't happen, and the sink should be released when a different source requests playback.