volumio/Volumio2 #1850: Make logging system configurable

layereight, 2019-11-24:
# Observation
* volumio uses winston as logging system
* in https://github.com/volumio/Volumio2/blob/master/app/index.js#L29 winston is configured to log
  * to the console (which is picked up by systemd's journal)
  * to the file `/var/log/volumio.log`
* as far as I can see "info" is the default logging level
* on "info" volumio seems to be quite verbose

# Motivation
* when you use a Raspberry Pi this has 2 implications
  * your SD card wears off quicker because relatively much is written to the file system
  * CPU usage is higher (you can actually see `systemd-journald` spike when it receives logs from volumio)
* I'm using a Raspberry Pi Zero powered by a power bank (which works great with volumio btw), so energy consumption is rather critical to me

# Workaround
* I couldn't find any way (or am I just missing something?) to configure logging level or targets
* as a workaround I added a line to https://github.com/volumio/Volumio2/blob/master/app/index.js
```
this.logger.level = 'error';
```
* right after the logger is defined

# Goal
* I'd like to be able to configure logging targets (files, console, ...) and logging levels comfortable, e.g.
  * in a configuration file
  * in the volumio UI
  * ...

ashthespy, 2019-11-25:
>your SD card wears off quicker because relatively much is written to the file system 

AFIK, the journal files aren't written to SD directly, but cached in the RAM..

volumio, 2019-11-27:
IMHO logging adds a neglectible amount of CPU usage. And as already stated, it does not impact SD card (all logggin is done on RAM, max 20MB used)

layereight, 2019-12-12:
You are right. journald is configured with `Storage=volatile` and `/var/log` is a `tmpfs`. So the SD card is indeed not impacted. Sorry, I missed that.

Usually, I'd agree that logging can be neglected when it comes to CPU usage. But unfortunatel not in my case. I trigger volumio's REST API on a Raspberry Pi Zero. I start a playlist to be precise. I end up having up to 250 lines of logs in systemd's journal and in `/var/log/volumio.log`. Most of it looks like actual debug logging at INFO log level. All those logs delay the playback on my Raspberry Pi Zero. It is definitely sensible, otherwise I wouldn't come up with this issue. Changing the log level to `error` makes a big difference. The delay is shorter then.

Maybe it's an edge case for volumio since the Raspberry Pi Zero doesn't have as much compute power as other systems. Plus I want to save as much energy as possible because my Pi Zero runs on battery.

Still, I think being able to configure volumio's logging system is a good idea. And no offense, who needs all that logging on a production environment anyway?