volumio/Volumio2 #1851: Webradio Stream Watchdog

Playasol, 2019-11-25:
Hi, when internet stream goes down or loses internet connection playback completly stops and it does not resume. I need a feature that constantly retries to play because i leave the device unatended, a simple ping to the radio server that loops stop and play until it recieves response would be a great solution. I tried to implement the mpd-watchdog service but I couldn't get it to work. Can I get a fix on this or some advice? Thank You for reading. : )

volumio, 2019-11-27:
Hi, this is out of the scope of Volumio. But we've implemented it in a new service we are about to launch ;)

Playasol, 2020-08-03:
> Hi, this is out of the scope of Volumio. But we've implemented it in a new service we are about to launch ;)

What is this service you mentioned? Im veru interested.