volumio/Volumio2 #1852: after update to 2.671 webradio not working

DerMatze87, 2019-11-26:
After I updated my raspberry to volume 2.671, web radio doesn't work. I can choose a radio station and push play button, but nothing will happen. A part of the log is here:

> Nov 26 23:39:36 volumio volumio[982]: info: CoreCommandRouter::volumioPlay
> Nov 26 23:39:36 volumio volumio[982]: UNSET VOLATILE
> Nov 26 23:39:36 volumio volumio[982]: info: CoreStateMachine::play index undefined
> Nov 26 23:39:36 volumio volumio[982]: info: CoreStateMachine::setConsumeUpdateService undefined
> Nov 26 23:39:36 volumio volumio[982]: info: CorePlayQueue::getTrack 9
> Nov 26 23:39:36 volumio volumio[982]: info: CoreStateMachine::startPlaybackTimer
> Nov 26 23:39:36 volumio volumio[982]: info: CorePlayQueue::getTrack 9
> Nov 26 23:39:36 volumio volumio[982]: info: [1574811576571] ControllerWebradio::clearAddPlayTrack
> Nov 26 23:39:36 volumio volumio[982]: info: ControllerMpd::sendMpdCommand stop

What's the reason for this issue? Thank you!

xipmix, 2019-12-03:
Does this happen with any webradio - sounds like that is the case. Can you list one or two you have tried here, to help reproduce?

I tried 2.673, released on 1 December. 'Hitradio O3' works ok. Can you try that version and let us know how you get on?

DerMatze87, 2019-12-04:
This error happens with any webradio, I tried "Inselradio Mallorca", "MDR Sputnik" and a lot of more. Now I'm on version 2.673 and everything is fine.

xipmix, 2019-12-05:
Cool, if it is fixed could you please close this issue.

closed 2019-12-10 by DerMatze87