volumio/Volumio2 #1854: Can't play some Qobuz albums

pelicangris, 2019-12-01:
Hi,

I have left a post on the Volumio forum but github is probably more appropriate for my issue.

I am using a python script to make socketio API call to volumio to play some tracks, e.g:
self.socket.emit('replaceAndPlay', {'uri': 'qobuz://myalbums/xxxxxxxxxxx', 'service': 'streaming_services'})

Most of the time this works fine but for some tracks, this fails with the below error:

info: An error occurred while exploding URI: TypeError: Cannot read property 'samplerate' of null

Full logs are available here:
http://logs.volumio.org/volumio/dr4a6xl.html

I have had a brief look at the code but my coding skills are long gone...
Any help would be appreciated!

xipmix, 2019-12-03:
Is it possible to look at those tracks with some other tool to see what metadata they have? I assume they all have a 'samplerate' field but it won't hurt to confirm.

It may help debugging to uncomment this line: https://github.com/volumio/Volumio2/blob/d5cf21d490b03385e906d8902aabafb6020783bc/app/playqueue.js#L197
and try to play one of the troublesome tracks.

pelicangris, 2019-12-29:
hey,
apologies for the lack of reply, i did miss your answer.
i have worked around it by using an alternate qobuz album id, they appear to be listing several versions of the album and i think some of them are available in hi-res only. As i only have access to standard quality, this particular one was failing.