volumio/Volumio2 #1862: Volumio constantly resolving mqtt.volumio.org

reshnm, 2019-12-18:
Hi,

my raspi volumio instance (2.657) is doing a DNS lookup for "mqtt.volumio.org" 3 times each second.
Is there some reason this is done? Seems to be a bit chatty ...

Rene

volumio, 2019-12-19:
mqtt.volumio.org is our broker for mqtt communication, used for myvolumio presence system and in the future for push updates.
Don't know why it resolves 3 times a second, the mqtt protocol should keep the connection alive with minimum bandwith and request

WarmChocolateCake, 2019-12-30:
Is there a way of disabling this? I'm happy to use my instance at home and have no plans for remote connectivity at the moment
(If I change my mind, then I'd enable it in the future :))

volumio, 2020-03-12:
Sorry not at the moment

closed 2020-03-12 by volumio

WarmChocolateCake, 2020-03-17:
OK, I'll just have to block Volumio from the internet.

volumio, 2020-03-17:
@WarmChocolateCake mind that some functions might not work properly (webradios, myvolumio, updater ).
Let me know what does not work after you disable it.
We now introduced privacy settings, this could be a good candidate to end up there

WarmChocolateCake, 2020-03-17:
That said... just doing a packet capture now (version 2.703) and I'm not even seeing any MQTT requests, so in my case, I'm not seeing the OP's high traffic.

WarmChocolateCake, 2020-03-17:
OK, comments passing each other. I'm going to do an update to the latest version and see what's happening

volumio, 2020-03-17:
The only reason why there can be lots of MQTT packets is when our MQTT broker was down, it happened few times between december and january...