volumio/Volumio2 #1874: After bcm wl was maked install,the interface is eth1,not wlan0

wghemail, 2020-01-05:
After bcm wl was maked install,the interface is eth1,not wlan0 .
I need modify /etc/udev/rules.d/75-persistent-net-generator.rules,change interface name to "wlan0"

volumio, 2020-01-08:
@wghemail can you give more info? which device becomes eth1?
which mods did you apply to net generator rules?
@macmpi FYI

xipmix, 2020-01-08:
A log showing this happening would be a help (https://volumio.github.io/docs/Development_How_To/Debugging.html). It might be best if you could undo your change to 75-persistent-net-generator.rules, collect the log and then put your change back.

macmpi, 2020-01-08:
@wghemail looking at your github, it seems you are doing some work on x86/5.4/Buster.
More context info appreciated indeed.

Anyhow, please be aware that starting with Stretch, default network interface naming convention has changed, but Volumio is still built on older conventions.
So you need to take care of forcing old naming conventions [as indicated here](https://github.com/volumio/Build/issues/259#issuecomment-328305239).