volumio/Volumio2 #1877: 3 letter artists not shown in the Artist list

monte-monte, 2020-01-08:
The problem is that though MPD scans and accepts artists with only 3 letters Volumio doesn't show them in list. You can search them, you can open them by album, but you don't see them in list.
That's not a big issue, because you can search them, but it's inconvenient if they contain special regional character, which you can't type with English keyboard layout. For example icelandic band called "múm".

xipmix, 2020-01-08:
This is odd. I have a test NAS share ('music library') with 'U2' as the artist name and it works fine.

My system is 2.692 on rpi2. What version are you running please?
Can you display the output of 'mediainfo <musicfile>' or similar so we can see what metadata mpd has to go on?

monte-monte, 2020-01-08:
@xipmix sorry, my bad. It is displayed. The isuue is that Volumio sorts artists not just alphabetically but also by letter case. So lower case is displayed in the end of the list. Don't know if it is intended, or bug, but certainly inconvenient.

xipmix, 2020-01-09:
Ah I see. I am not sure what the best behaviour is. Most likely defaulting to case-insensitive is correct here because most  Artists, being so big and important, like to have their names Captalised. Could you retitle this please - something like: please sort Artist lists case-insensitively ?