volumio/Volumio2 #1878: WebUI not responsive when no internet connection available

joshwapohlmann, 2020-01-08:
Hello,

I installed volumio 2.692, configured the device via the volumio AP to connect itself to my wifi network.
This wifi network has **NO** internet connection.
When booted up, everything works, I can SSH into the device **BUT** the WebUI remains unresponsive.

1. What kind of logs do I have to review to find the cause?
So far, I only logged with `journalctl -f`
2. The only thing out of the ordinary is "Cannot call home". Does this block the startup?
3. Does volumio expect connectivity over eth0 and therefore waits until it is established? Does it fail to recognize wlan0 as the main interface since it does not provide internet connectivity?

Here the normal log with an unresponsive WebUI
```
-- Logs begin at Mi 2020-01-08 10:14:09 UTC. --
Jan 08 10:14:40 biotspeaker volumio[1014]: info: CorePlayQueue::getTrack 0
Jan 08 10:14:40 biotspeaker sudo[1192]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 08 10:14:40 biotspeaker volumio[1014]: info: CoreCommandRouter::executeOnPlugin: multiroom , getMultiroom
Jan 08 10:14:40 biotspeaker volumio[1014]: info: Listing playlists
Jan 08 10:14:40 biotspeaker systemd[1]: Starting UPnP Renderer front-end to MPD...
Jan 08 10:14:40 biotspeaker systemd[1]: Started UPnP Renderer front-end to MPD.
Jan 08 10:14:40 biotspeaker sudo[1192]: pam_unix(sudo:session): session closed for user root
Jan 08 10:14:40 biotspeaker volumio[1014]: info: Upmpdcli Daemon Started
Jan 08 10:14:41 biotspeaker sudo[1220]: volumio : TTY=pts/0 ; PWD=/home/volumio ; USER=root ; COMMAND=/bin/journalctl -f
Jan 08 10:14:41 biotspeaker sudo[1220]: pam_unix(sudo:session): session opened for user root by volumio(uid=0)

Jan 08 10:14:42 biotspeaker systemd-timesyncd[332]: Using NTP server 192.168.178.1:123 (fritz.box).

Jan 08 10:14:42 biotspeaker volumio-remote-updater[567]: [2020-01-08 10:14:42] [connect] Successful connection
Jan 08 10:14:42 biotspeaker volumio-remote-updater[567]: [2020-01-08 10:14:42] [connect] WebSocket Connection 127.0.0.1:3000 v-2 "WebSocket++/0.5.1" /socket.io/?EIO=4&transport=websocket&t=1578478482 101

Jan 08 10:14:43 biotspeaker ifup[423]: dhcpcd[458]: timed out
Jan 08 10:14:43 biotspeaker dhcpcd[458]: timed out
Jan 08 10:14:43 biotspeaker dhcpcd[458]: exited
Jan 08 10:14:43 biotspeaker ifup[423]: dhcpcd[458]: exited
Jan 08 10:14:43 biotspeaker ifup[423]: Failed to bring up eth0.
Jan 08 10:14:52 biotspeaker volumio[1014]: info: CoreCommandRouter::volumioGetState
Jan 08 10:14:52 biotspeaker volumio[1014]: info: CoreStateMachine::getState
Jan 08 10:14:52 biotspeaker volumio[1014]: info: CorePlayQueue::getTrack 0
Jan 08 10:14:52 biotspeaker volumio[1014]: info: CoreCommandRouter::executeOnPlugin: multiroom , getMultiroom
Jan 08 10:14:52 biotspeaker volumio[1014]: info: Listing playlists
Jan 08 10:14:53 biotspeaker systemd-timesyncd[332]: Timed out waiting for reply from 192.168.178.1:123 (fritz.box).
Jan 08 10:14:56 biotspeaker ntpd_intres[685]: host name not found: 0.debian.pool.ntp.org
Jan 08 10:15:02 biotspeaker volumio[1014]: info: CoreCommandRouter::executeOnPlugin: system , getSystemVersion
Jan 08 10:15:02 biotspeaker volumio[1014]: info: CoreCommandRouter::executeOnPlugin: system , getConfigParam
Jan 08 10:15:06 biotspeaker volumio[1014]: info: CoreCommandRouter::volumioGetState
Jan 08 10:15:06 biotspeaker volumio[1014]: info: CoreStateMachine::getState
Jan 08 10:15:06 biotspeaker volumio[1014]: info: CorePlayQueue::getTrack 0
Jan 08 10:15:06 biotspeaker volumio[1014]: Discovery: adding fe6a8ad3-f078-481d-86aa-4ba147a9a2c5
Jan 08 10:15:06 biotspeaker volumio[1014]: info: mDNS: Found device biotspeaker
Jan 08 10:15:06 biotspeaker volumio[1014]: info:
Jan 08 10:15:06 biotspeaker volumio[1014]: ---------------------------- Client requests Volumio Play Playlist HA_hibernating
Jan 08 10:15:06 biotspeaker sudo[1283]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/sbin/ifconfig wlan0
Jan 08 10:15:06 biotspeaker volumio[1014]: info: Play playlist HA_hibernating
Jan 08 10:15:06 biotspeaker sudo[1283]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 08 10:15:07 biotspeaker sudo[1283]: pam_unix(sudo:session): session closed for user root
Jan 08 10:15:16 biotspeaker volumio[1014]: info: Cannot call home: Error: Command failed: /usr/bin/curl -X POST --data-binary "device=pi&variante=volumio&version=2.692&uuid=48a03a5e9a820d1087fe38c45552c2de" http://updates.volumio.org:7070/downloader-v1/track-device
Jan 08 10:15:16 biotspeaker volumio[1014]: % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
Jan 08 10:15:16 biotspeaker volumio[1014]: Dload  Upload   Total   Spent    Left  Speed
Jan 08 10:15:16 biotspeaker volumio[1014]: [1.9K blob data]
Jan 08 10:15:16 biotspeaker volumio[1014]: [291B blob data]
Jan 08 10:15:16 biotspeaker volumio[1014]: retrying in 5 seconds, trial 0
Jan 08 10:15:16 biotspeaker volumio[1014]: Volumio Calling Home
Jan 08 10:15:24 biotspeaker ntpd_intres[685]: host name not found: 1.debian.pool.ntp.org
Jan 08 10:15:34 biotspeaker volumio[1014]: info: Listing playlists
Jan 08 10:15:34 biotspeaker volumio[1014]: info: Listing playlists
Jan 08 10:15:52 biotspeaker ntpd_intres[685]: host name not found: 2.debian.pool.ntp.org
Jan 08 10:15:55 biotspeaker volumio[1014]: info: Cannot call home: Error: Command failed: /usr/bin/curl -X POST --data-binary "device=pi&variante=volumio&version=2.692&uuid=48a03a5e9a820d1087fe38c45552c2de" http://updates.volumio.org:7070/downloader-v1/track-device
Jan 08 10:15:55 biotspeaker volumio[1014]: % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
Jan 08 10:15:55 biotspeaker volumio[1014]: Dload  Upload   Total   Spent    Left  Speed
Jan 08 10:15:55 biotspeaker volumio[1014]: [1.9K blob data]
[... 82 lines cut ...]
Jan 08 10:16:15 biotspeaker volumio[1014]: info: CoreCommandRouter::volumioGetState
Jan 08 10:16:15 biotspeaker volumio[1014]: info: CoreStateMachine::getState
Jan 08 10:16:15 biotspeaker volumio[1014]: info: CorePlayQueue::getTrack 0
Jan 08 10:16:15 biotspeaker sudo[1640]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/sbin/ifconfig wlan0
Jan 08 10:16:15 biotspeaker sudo[1640]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 08 10:16:15 biotspeaker volumio[1014]: info: CoreCommandRouter::executeOnPlugin: multiroom , getMultiroom
Jan 08 10:16:15 biotspeaker volumio[1014]: info: Listing playlists
Jan 08 10:16:15 biotspeaker sudo[1640]: pam_unix(sudo:session): session closed for user root
Jan 08 10:16:15 biotspeaker sudo[1644]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/sbin/ifconfig wlan0
Jan 08 10:16:15 biotspeaker sudo[1644]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 08 10:16:15 biotspeaker sudo[1644]: pam_unix(sudo:session): session closed for user root
Jan 08 10:16:17 biotspeaker systemd-timesyncd[332]: Using NTP server 213.172.105.106:123 (3.debian.pool.ntp.org).
Jan 08 09:32:24 biotspeaker systemd[1055]: Time has been changed
Jan 08 09:32:24 biotspeaker systemd-timesyncd[332]: interval/delta/delay/jitter/drift 32s/-2632.797s/0.015s/0.000s/+0ppm
Jan 08 09:32:24 biotspeaker systemd[1]: Time has been changed
Jan 08 09:32:24 biotspeaker volumio[1014]: info: Fetching Streaming Services browse cache
Jan 08 09:32:24 biotspeaker volumio[1014]: info: CoreCommandRouter::volumioRemoveToBrowseSourcesTIDAL
Jan 08 09:32:24 biotspeaker volumio[1014]: info: CoreCommandRouter::executeOnPlugin: my_music , getDisabledSources
Jan 08 09:32:24 biotspeaker volumio[1014]: info: CoreCommandRouter::volumioRemoveToBrowseSourcesQOBUZ
Jan 08 09:32:24 biotspeaker volumio[1014]: info: CoreCommandRouter::executeOnPlugin: my_music , getDisabledSources
Jan 08 09:32:27 biotspeaker ntpd_intres[685]: host name not found: 3.debian.pool.ntp.org
Jan 08 09:32:28 biotspeaker volumio[1014]: info: CoreCommandRouter::volumioGetState
Jan 08 09:32:28 biotspeaker volumio[1014]: info: CoreStateMachine::getState
Jan 08 09:32:28 biotspeaker volumio[1014]: info: CorePlayQueue::getTrack 0
```

xipmix, 2020-01-09:
Something like this has been seen before (#1680, #1588. #1456), it would be good to get you going and document what to do.

```journalctl``` has most of what you will need. Also check ```/var/log/volumio.log```.

Let me see if I understand your setup steps
 * you used the 'volmio' hotspot for initial setup
 * you set up connection to your home wifi
 * you left the wired ethernet unplugged

How am I doing?

Questions:
 * Is the 'hotspot' still enabled? It shouldn't be needed if the wifi interface is working.
 * if you disable the wired interface (Cog->Network), do things improve?
 * If you disable albumart fetching (Cog->Appearance(?)), do things improve?

joshwapohlmann, 2020-01-09:
> you used the 'volmio' hotspot for initial setup

yes

> you set up connection to your home wifi

yes

> you left the wired ethernet unplugged

yes

you do well.

> Is the 'hotspot' still enabled? It shouldn't be needed if the wifi interface is working.

yes it was, i disabled it

> if you disable the wired interface (Cog->Network), do things improve?
> If you disable albumart fetching (Cog->Appearance(?)), do things improve?

I did all of them and restarted the device. The WebUI now loads in parts. takes a long time to load and sometimes doesn't response. The logs say NOTHING about this. They only log the commands issues by the API/Webinterface. Can I somehow debug this?

I did some further testing. It is not about ethernet or wifi. It is about internet connectivity. When I provide connectivity to the network, the webinterface SUDDENLY works. 

See this video:
[image]
Here you see the moment volumio gets connection. Before that the loading sometimes doesn't even start or hangs (maybe just pulling from cache). 

See the logs. Note the moment NTP gets a connection to the servers. That moment volumio also starts working. I don't know why this is linked to internet connectivity.
```
Jan 09 14:14:41 biotspeaker ntpd_intres[697]: DNS 1.debian.pool.ntp.org -> 213.239.239.164
Jan 09 14:14:41 biotspeaker ntpd_intres[697]: DNS 2.debian.pool.ntp.org -> 212.18.3.19
Jan 09 14:14:41 biotspeaker ntpd_intres[697]: DNS 3.debian.pool.ntp.org -> 176.9.241.107
Jan 09 14:14:42 biotspeaker volumio[1058]: info: [MyVolumio PluginManager] Adding plugin bluetooth to plan superstar
Jan 09 14:14:42 biotspeaker volumio[1058]: info: [MyVolumio PluginManager] Adding plugin cd_controller to plan superstar
Jan 09 14:14:42 biotspeaker volumio[1058]: info: [MyVolumio PluginManager] Adding plugin smart_inputs to plan superstar
Jan 09 14:14:42 biotspeaker volumio[1058]: info: [MyVolumio PluginManager] Adding plugin hi_res_audio to plan superstar
Jan 09 14:14:42 biotspeaker volumio[1058]: info: [MyVolumio PluginManager] Adding plugin bluetooth to plan virtuoso
Jan 09 14:14:42 biotspeaker volumio[1058]: info: [MyVolumio PluginManager] Adding plugin cd_controller to plan virtuoso
Jan 09 14:14:42 biotspeaker volumio[1058]: info: Adding plugin bluetooth to MyMusic Plugins
Jan 09 14:14:42 biotspeaker volumio[1058]: info: Adding plugin cd_controller to MyMusic Plugins
Jan 09 14:14:42 biotspeaker volumio[1058]: info: Adding plugin smart_inputs to MyMusic Plugins
Jan 09 14:14:42 biotspeaker volumio[1058]: info: [MyVolumio PluginManager] Loading plugin "my_volumio"...
Jan 09 14:14:46 biotspeaker volumio[1058]: info: [MyVolumio PluginManager] Loading plugin "streaming_services"...
Jan 09 14:14:49 biotspeaker volumio[1058]: info: [MyVolumio PluginManager] MyVolumio plugin successfully loaded
Jan 09 14:14:49 biotspeaker volumio[1058]: info: [MyVolumio PluginManager] Starting plugin system_controller.my_volumio
Jan 09 14:14:49 biotspeaker volumio[1058]: info: CoreCommandRouter::executeOnPlugin: system , getConfigParam
Jan 09 14:14:49 biotspeaker volumio[1058]: info: CoreCommandRouter::executeOnPlugin: system , getConfigParam
Jan 09 14:14:49 biotspeaker volumio[1058]: info: Starting MyVolumio Remote Streaming Endpoints
Jan 09 14:14:50 biotspeaker volumio[1058]: info: [MyVolumio PluginManager] Streaming Services plugin successfully loaded
Jan 09 14:14:50 biotspeaker volumio[1058]: info: [MyVolumio PluginManager] Starting plugin music_service.streaming_services
Jan 09 14:14:50 biotspeaker volumio[1058]: info: Streaming services startup
Jan 09 14:14:50 biotspeaker volumio[1058]: info: Starting Streaming Daemon
Jan 09 14:14:50 biotspeaker volumio[1058]: info: MyVolumio not started
Jan 09 14:14:50 biotspeaker volumio[1058]: info: Initializing device activation check
Jan 09 14:14:50 biotspeaker sudo[1902]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl restart volumio-streaming-daemon.service
Jan 09 14:14:50 biotspeaker sudo[1902]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 09 14:14:50 biotspeaker volumio[1058]: info: [MyVolumio PluginManager] MyVolumio plugin successfully started
Jan 09 14:14:50 biotspeaker systemd[1]: Stopping Volumio Streaming Daemon...
Jan 09 14:14:50 biotspeaker systemd[1]: Starting Volumio Streaming Daemon...
Jan 09 14:14:50 biotspeaker systemd[1]: Started Volumio Streaming Daemon.
Jan 09 14:14:50 biotspeaker sudo[1902]: pam_unix(sudo:session): session closed for user root
Jan 09 14:14:50 biotspeaker volumio-streaming-daemon[1908]: ############################
Jan 09 14:14:50 biotspeaker volumio-streaming-daemon[1908]: # Volumio Streaming Daemon #
Jan 09 14:14:50 biotspeaker volumio-streaming-daemon[1908]: # Running on port 7777     #
Jan 09 14:14:50 biotspeaker volumio-streaming-daemon[1908]: ############################
....
```

xipmix, 2020-01-10:
Something that frequently causes jumpy behaviour and lockups is DNS.
Have you tried setting a custom DNS server (Cog->Network->Custom DNS) yet?

Perhaps try this command, to get an idea of what sequence of services volumio.service needs to start
```
$ sudo systemd-analyze critical-chain volumio.service
```
If you try this after you have let volumio boot without network and sit for a while (say 1m) you should see a large time step on the one that was holding things up.

It was interesting that NTP was the first thing that logged anything.
What if you disable NTP and reboot (not just a volumio restart)
```
$ sudo systemctl mask ntp; sudo reboot
```
I tried this on my (internet - connected) volumio and didn't see any immediate issues.
This is just a test - you will need to do something about time service as the time will drift if left unchecked.

joshwapohlmann, 2020-01-10:
Thanks for your ongoing support!

> $ sudo systemd-analyze critical-chain volumio.service

Thanks for this new command I didn't know about before!
I didn't reboot the system after I made the video above. So the chain you see here is the exact same from the "proof" video:
```
volumio.service @25.270s
└─wireless.service @5.725s +19.539s
  └─basic.target @5.626s
    └─timers.target @5.618s
      └─systemd-tmpfiles-clean.timer @5.618s
        └─sysinit.target @5.616s
          └─systemd-rfkill@rfkill1.service @7.330s +74ms
            └─system-systemd\x2drfkill.slice @3.042s
              └─system.slice @525ms
                └─-.slice @522ms
```
This tree looks the same when rebooted and internet connectivity hasn't been provided yet. So it doesn't provide any more information unfortunately.

> It was interesting that NTP was the first thing that logged anything.
> What if you disable NTP and reboot (not just a volumio restart)

I enabled NTP manually. It is not by default in volumio. I achieved that by enabling the servers in `/etc/systemd/timesyncd.conf` and added my local FritzBox (router/Wifi AP). So that's why you would normally don't see this in your logs.
```
[Time]
Servers=fritz.box 0.debian.pool.ntp.org 1.debian.pool.ntp.org 2.debian.pool.ntp.org 3.debian.pool.ntp.org
```

I compared log entries before connectivity and after. The following blob is definitly the interesting part.
```
Jan 10 10:54:27 biotspeaker volumio[1054]: info: [MyVolumio PluginManager] Adding plugin bluetooth to plan superstar
Jan 10 10:54:27 biotspeaker volumio[1054]: info: [MyVolumio PluginManager] Adding plugin cd_controller to plan superstar
Jan 10 10:54:27 biotspeaker volumio[1054]: info: [MyVolumio PluginManager] Adding plugin smart_inputs to plan superstar
Jan 10 10:54:27 biotspeaker volumio[1054]: info: [MyVolumio PluginManager] Adding plugin hi_res_audio to plan superstar
Jan 10 10:54:27 biotspeaker volumio[1054]: info: [MyVolumio PluginManager] Adding plugin bluetooth to plan virtuoso
Jan 10 10:54:27 biotspeaker volumio[1054]: info: [MyVolumio PluginManager] Adding plugin cd_controller to plan virtuoso
Jan 10 10:54:27 biotspeaker volumio[1054]: info: Adding plugin bluetooth to MyMusic Plugins
Jan 10 10:54:27 biotspeaker volumio[1054]: info: Adding plugin cd_controller to MyMusic Plugins
Jan 10 10:54:27 biotspeaker volumio[1054]: info: Adding plugin smart_inputs to MyMusic Plugins
Jan 10 10:54:27 biotspeaker volumio[1054]: info: [MyVolumio PluginManager] Loading plugin "my_volumio"
Jan 10 10:54:31 biotspeaker volumio[1054]: info: [MyVolumio PluginManager] Loading plugin "streaming_services"...
Jan 10 10:54:34 biotspeaker volumio[1054]: info: [MyVolumio PluginManager] MyVolumio plugin successfully loaded
Jan 10 10:54:34 biotspeaker volumio[1054]: info: [MyVolumio PluginManager] Starting plugin system_controller.my_volumio
Jan 10 10:54:34 biotspeaker volumio[1054]: info: CoreCommandRouter::executeOnPlugin: system , getConfigParam
Jan 10 10:54:34 biotspeaker volumio[1054]: info: CoreCommandRouter::executeOnPlugin: system , getConfigParam
Jan 10 10:54:34 biotspeaker volumio[1054]: info: Starting MyVolumio Remote Streaming Endpoints
Jan 10 10:54:34 biotspeaker volumio[1054]: info: [MyVolumio PluginManager] Streaming Services plugin successfully loaded
Jan 10 10:54:34 biotspeaker volumio[1054]: info: [MyVolumio PluginManager] Starting plugin music_service.streaming_services
Jan 10 10:54:34 biotspeaker volumio[1054]: info: Streaming services startup
Jan 10 10:54:34 biotspeaker volumio[1054]: info: Starting Streaming Daemon
Jan 10 10:54:34 biotspeaker volumio[1054]: info: MyVolumio not started
Jan 10 10:54:34 biotspeaker volumio[1054]: info: Initializing device activation check
Jan 10 10:54:34 biotspeaker volumio[1054]: info: [MyVolumio PluginManager] MyVolumio plugin successfully started
Jan 10 10:54:34 biotspeaker systemd[1]: Stopping Volumio Streaming Daemon...
Jan 10 10:54:34 biotspeaker systemd[1]: Starting Volumio Streaming Daemon...
Jan 10 10:54:34 biotspeaker systemd[1]: Started Volumio Streaming Daemon.
Jan 10 10:54:34 biotspeaker volumio-streaming-daemon[2406]: ############################
Jan 10 10:54:34 biotspeaker volumio-streaming-daemon[2406]: # Volumio Streaming Daemon #
Jan 10 10:54:34 biotspeaker volumio-streaming-daemon[2406]: # Running on port 7777     #
Jan 10 10:54:34 biotspeaker volumio-streaming-daemon[2406]: ############################
```

joshwapohlmann, 2020-01-10:
I tried out configuring the FritzBox as DNS server. Didn't help.
Since the source code of `MyVolumio PluginManager` seems to be closed source so I couldn't have a look whats causing the stall, I simply **deleted** this ... thing.
No, it works like a charm!

## Solution
Delete MyVolumio with `$ sudo rm -r /myvolumio`

closed 2020-01-10 by joshwapohlmann

volumio, 2020-01-10:
Interesting. This is not the intended behaviour. We will have a look.