volumio/Volumio2 #1881: API Feature Request: Call web radio via play

jwoodard80, 2020-01-17:
Please forgive me if this is the wrong place for this. Would it be possible to extend the play command to accept one of the web radio URLs? 

Currently the only workaround I know of is to create a playlist with only the web radio inside and call playplaylist. 

Were I more fluent in JS I wouldn't mind trying but I'm hoping it won't be a big change for someone. 

Thank you for the consideration and great project.

shed, 2020-01-27:
I would love to see this as well.

fr1sty, 2020-04-01:
You can using the url http://volumio.local/api/v1/replaceAndPlay
And then posting the data, in this case for wefunkradio :)
`
{
"service": "webradio",
"type": "webradio",
"title": "WEFUNK",
"uri": "http://s-09.wefunkradio.com:81/wefunk64.mp3"
}
`

jwoodard80, 2020-04-01:
This works very well! Thank you so much!

closed 2020-04-01 by jwoodard80