volumio/Volumio2 #1882: Broken volumio after update 2.698

damien122, 2020-01-17:
After I install the latest Update 2.698 my volumio did a reboot and afterwards nothing works. The web interface is not reachable and it don't play any music. I can connect to ssh and can see some node processes working but don't know what can I do.

xipmix, 2020-01-17:
You can run a log capture from ssh 
```
/usr/local/bin/volumio logdump
```
and upload it here, or pastebin.

damien122, 2020-01-18:
Here is my logdump. Also I noticed that I can hear the startup sound very quiet in a loop playing. Like volumio is starting again and again. But the system don't do a reboot because I am still connected to ssh.

[logondemand.txt](https://github.com/volumio/Volumio2/files/4080344/logondemand.txt)

kermitnc, 2020-01-19:
I had a similar issue. I restarted the rpi (3b+) twice by disconnecting the usb power line. Now it is working for me.

damien122, 2020-01-20:
@kermitnc didn't work for me. 

How can I do a factory reset from ssh?

volumio, 2020-01-20:
volumio updater factory

This error has been taken care, we are to release an update soon

damien122, 2020-01-20:
@volumio nice, but how can I install the update when I can't access the web ui?

volumio, 2020-01-20:
Can you let us know which plugins you have installed?

damien122, 2020-01-20:
Backup & Restore, Autoplay (one of the two) and Spotify Connect

damien122, 2020-01-24:
@volumio are there some news about this bug?

volumio, 2020-01-24:
New release is out, which should fix the issue. 
Pleasee confirm it does and close the issue in this case.

damien122, 2020-01-24:
And how should I install this fix when I can't access the frontend? I have just ssh access.

volumio, 2020-01-24:
You can reset to factory with

volumio updater factory

Then you will be able to update

damien122, 2020-01-24:
Sorry but I didn't understand how to call the "volumio updater factory"

volumio, 2020-01-24:
From ssh command line

damien122, 2020-01-25:
bug is fixed with release 2.699

closed 2020-01-25 by damien122