volumio/Volumio2 #1885: Volumio not functional since last update

gael1980, 2020-01-22:


xipmix, 2020-01-22:
This is version 2.698 which has a known issue. Look for a new release soon.

gael1980, 2020-01-23:
All right, thanks.

futuralogic, 2020-02-04:
Any follow up? Similar issue here.
My scenario: nfs Mount remains but library shows nothing. My library NAS is gone. 
Rescan doesn’t work.

xipmix, 2020-02-05:
Thanks for reporting. Are you on 2.699 (Released 20 Jan), or earlier? It may be that both 2.698 and 2.699 have this issue. See also #1887

futuralogic, 2020-02-05:
@xipmix My issue is same as #1887 
I'm planning to take time this evening to grab logs from my two devices that are experiencing this issue and posting info on #1887. I did just upgrade to 2.699. I don't recall version I was on before but believe it was the previous public version as I had been fairly up to date. I did not experience the problem under that previous version.

gael1980, 2020-02-05:
same thing, I had no problem with the version before 2.698

xipmix, 2020-02-06:
Just checking - you both upgraded using the OTA process (cog->system->check_updates) ?

volumio, 2020-02-06:
Can you post the content of /etc/mpd.conf ?

cat /etc/mpd.conf

gael1980, 2020-02-06:
# Volumio MPD Configuration File

# Files and directories #######################################################
music_directory		"/var/lib/mpd/music"
playlist_directory		"/var/lib/mpd/playlists"
db_file			"/var/lib/mpd/tag_cache"
log_file			"/var/log/mpd.log"
#pid_file			"/var/run/mpd/pid"
#state_file			"/var/lib/mpd/state"
#sticker_file                   "/var/lib/mpd/sticker.sql"
###############################################################################

# General music daemon options ################################################
user				"mpd"
group                          "audio"
bind_to_address		"any"
#port				"6600"
#log_level			"default"
gapless_mp3_playback			"no"
#save_absolute_paths_in_playlists	"no"
#metadata_to_use	"artist,album,title,track,name,genre,date,composer,performer,disc"
auto_update    "yes"
#auto_update_depth "3"
###############################################################################
# Symbolic link behavior ######################################################
follow_outside_symlinks	"yes"
follow_inside_symlinks		"yes"
###############################################################################
# Input #######################################################################
#
input {
        plugin "curl"
#       proxy "proxy.isp.com:8080"
#       proxy_user "user"
#       proxy_password "password"
}
###############################################################################

# Decoder ################################################################

 

 

###############################################################################

# Audio Output ################################################################

resampler {      
  		plugin "soxr"
  		quality "very high"
  		threads "1"
}

audio_output {
		type		"alsa"
		name		"alsa"
		device		"hw:1,0"
		dop			"no"
		
		format      "*:32:2"
		

}

audio_output {
    type            "fifo"
    enabled         "no"
    name            "multiroom"
    path            "/tmp/snapfifo"
    format          "44100:16:2"
}

#replaygain			"album"
#replaygain_preamp		"0"
volume_normalization		"yes"
###############################################################################

# MPD Internal Buffering ######################################################
audio_buffer_size		"4096"
buffer_before_play		"10%"
###############################################################################

# Resource Limitations ########################################################
#connection_timeout		"60"
max_connections			"20"
max_playlist_length		"81920"
max_command_list_size		"81920"
max_output_buffer_size		"81920"
###############################################################################

# Character Encoding ##########################################################
filesystem_charset		"UTF-8"
id3v1_encoding			"UTF-8"
###############################################################################

volumio, 2020-02-06:
It seems all fine (just wondering if your output device is able to play at 32 bit as per your configuration).
Do you have other plugins installed?
Which hardware are you running?

gael1980, 2020-02-06:
I use an rpi3 + ALLO KALI Reclocker + Allo piano 2.1

I have to reconfigure the output sound otherwise I don't have an output sound. Well yes, but a white noise.

As I explain it, before the 2.698 update I had no problem.

volumio, 2020-02-06:
@gael1980 thanks for the explanation

futuralogic, 2020-02-06:
> Just checking - you both upgraded using the OTA process (cog->system->check_updates) ?

Yes.

xipmix, 2020-02-16:
I think this is fixed in v2.703, released 15-02-2020. Please close if that's the case.

robertgosz, 2020-03-02:
It's still broken on 2.713 (RPI3) 
First NAS scaning crashes. 
Library is empty. 
My music (sources) shows nothing. 
Update and rescan do not work.

xipmix, 2020-03-02:
Can I clarify please - this was after upgrading using the OTA process to 2.713, not a fresh install?
If you can ssh in to the machine, does the NAS get mounted (df will show a networked mount).
A log covering from bootup to a failed update/rescan would really help.

robertgosz, 2020-03-03:
Fresh 2.713 install on RPI3. Today - updated to 2.714. Still broken.
Logs attached (boot, add NAS volume, scan, rescan, update).
NAS is mounted and accessible - can be browsed via ssh.

[mpd.log](https://github.com/volumio/Volumio2/files/4282999/mpd.log)
[volumio.log](https://github.com/volumio/Volumio2/files/4283000/volumio.log)

[image]

volumio, 2020-03-03:
Please go to Playback settings and save the audio output (this will rewrite mpd.conf).
Let us know

robertgosz, 2020-03-03:
Looks like it made a difference. After the restart I could see a few tracks in My Music, but this time the update worked and did not crash after a minute or so. It's still scanning but I can already see over 11K tracks. Will do a bit more testing, but looks promising. Thanks!

closed 2020-03-12 by volumio