volumio/Volumio2 #1887: New issue since last update 2.699

gael1980, 2020-01-27:
Hello,
Since last update, my database is empty...
yet, my nas and flash drive are well connected...
when I click on "refresh database" volumio reboot
[image]

volumio, 2020-01-27:
Can you send a log after you press refresh ? 
See: https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html

Paste the link to logs here

gael1980, 2020-01-27:
http://logs.volumio.org/volumio/1ifEWpv.html

volumio, 2020-01-27:
Which plugins do you have installed?

gael1980, 2020-01-27:
Sysinfo
Spotify
spotify connect 2
Backup and restore
radio paradise

in the log file, i read one fatal error : 
Jan 27 13:58:27 volumio volumio[981]: TypeError: Cannot read property 'then' of undefined
Jan 27 13:58:27 volumio volumio[981]: at ControllerMpd.sendMpdCommand (/volumio/app/plugins/music_service/mpd/index.js:347:10)

gael1980, 2020-01-27:
here's another log
http://logs.volumio.org/volumio/ahzI4xX.html

I have a lot of : error: MPD error: Error: connect EAGAIN /run/mpd/socket

cbu99, 2020-02-02:
Same here on two different installations after update to 2.699.
error: MPD error: Error: connect EAGAIN /run/mpd/socket

gael1980, 2020-02-05:
Always the same error : 
2020-02-05T12:21:54.636Z - error: Cannot start MPD on system Start: Error: Command failed: /usr/bin/sudo /bin/systemctl restart mpd.service
Job for mpd.service failed. See 'systemctl status mpd.service' and 'journalctl -xn' for details.

futuralogic, 2020-02-05:
Confirming the same symptoms.

My logs:
Device 1: http://logs.volumio.org/volumio/B48Uamj.html
Device 2: http://logs.volumio.org/volumio/aLhQnw0.html

volumio, 2020-02-05:
Seems that mpd is failing to start.
Could you execute 
systemctl status mpd -l

cbu99, 2020-02-05:
volumio@volumiopi2:~$ systemctl status mpd -l
● mpd.service - Music Player Daemon
   Loaded: loaded (/lib/systemd/system/mpd.service; enabled)
   Active: activating (start) since Wed 2020-02-05 16:07:39 UTC; 54s ago
     Docs: man:mpd(1)
           man:mpd.conf(5)
 Main PID: 1764 (mpd)
   CGroup: /system.slice/mpd.service
           └─1764 /usr/bin/mpd --no-daemon

Feb 05 16:07:39 volumiopi2 systemd[1]: Unit mpd.service entered failed state.
Feb 05 16:07:39 volumiopi2 systemd[1]: Starting Music Player Daemon...
volumio@volumiopi2:~$ systemctl status mpd -l
● mpd.service - Music Player Daemon
   Loaded: loaded (/lib/systemd/system/mpd.service; enabled)
   Active: deactivating (final-sigterm) (Result: timeout) since Sun 2020-02-02 15:16:38 UTC; 3 days ago
     Docs: man:mpd(1)
           man:mpd.conf(5)
 Main PID: 1764 (mpd)
   CGroup: /system.slice/mpd.service
           └─1764 /usr/bin/mpd --no-daemon

Feb 05 16:07:39 volumiopi2 systemd[1]: Unit mpd.service entered failed state.
Feb 05 16:07:39 volumiopi2 systemd[1]: Starting Music Player Daemon...
Feb 05 16:09:09 volumiopi2 systemd[1]: mpd.service start operation timed out. Terminating.

gael1980, 2020-02-06:
Same error...

xipmix, 2020-02-06:
The logs say the startup process is timing out. I wonder if these might be helpful to spot what is taking too long. 
```
$ sudo systemd-analyze critical-chain mpd.service
$ sudo systemd-analyze critical-chain mpd.socket
```
Initially I was wondering if the NAS mount was taking too long (I am seeing some wierdness with NFS mounts) but your log shows the NAS is mounted (via NFS) just fine and before mpd tries to start.

This is strange, in mpd.log.
```
exception: Failed to register /var/lib/mpd/music/NAS/Bangkok/#recycle: inotify_add_watch() has failed: Permission denied
```
Can you have a look at the permissions in that tree? This is what I get
```
$ ls -ld /var/lib/mpd
drwxr-xr-x 1 mpd audio 4096 Dec 21 05:08 /var/lib/mpd
$ ls -ld /var/lib/mpd/music/                           
drwxr-xr-x 2 root root 71 Dec 19 14:25 /var/lib/mpd/music/
$ ls -ld /var/lib/mpd/music/NAS
lrwxrwxrwx 1 root root 8 Dec 19 14:25 /var/lib/mpd/music/NAS -> /mnt/NAS
$ ls -ld /var/lib/mpd/music/NAS/  #trailing slash resolves the symlink
drwxrwxrwx 1 root root 4096 Dec 21 05:09 /var/lib/mpd/music/NAS/
$ ls -l /var/lib/mpd/music/NAS/
total 4
drwxrwsr-x 8 volumio 1002 4096 Sep  1  2018 test  #in your case this will be 'Bankok'
```

volumio, 2020-02-06:
This can be a good hint. It might be that:
- If NAS is mounted after mpd starts, its all fine (as inotify will not add watches on start)
- If Nas is mounted, and mpd is restarted after that, it times out (inotify will cause the time out).

Can you try unmounting your Nas, then restarting the system and see if that works?

gael1980, 2020-02-06:
if I remove the link to the nas, it works, I only have in the library the musics of my USB key.
It also works when I reconnect my nas. The database is well updated.
On the other hand, it is not practical. My NAS is permanently connected to the network.

volumio, 2020-02-06:
Thanks gael, this finally explains the issue.
Looking for a fix as we speak

@xipmix what do you suggest?

gael1980, 2020-02-06:
maybe in order of starting services?

xipmix, 2020-02-06:
I'm confused. Am I to understand that once mpd starts and the NAS is mounted, then if mpd is restarted _at any time after that_, inotify is going to timeout? That seems like a bug in mpd. Or are we just trying to get the boot-time ordering right?

If the latter then on the face of it we want to force systemd to order remote-fs.service after mpd.service (but not too far after...). But are there other things that want remote-fs.service to start earlier?

futuralogic, 2020-02-07:
I am not sure this is relevant after the last couple posts, but here is the output of ls commands as you alluded above.
One thing I'd point out is that my NAS is mounted via NFS and is set as read-only. I haven't typically faced an issue with that, but could be the reason for the inotify errors in the logs I posted.

```
ls -ld /var/lib/mpd
drwxr-xr-x 1 mpd audio 4096 Dec  2 22:29 /var/lib/mpd
ls -ld /var/lib/mpd/music/
drwxr-xr-x 2 root root 71 Jan 20 18:41 /var/lib/mpd/music/
ls -ld /var/lib/mpd/music/NAS
lrwxrwxrwx 1 root root 8 Jan 20 18:41 /var/lib/mpd/music/NAS -> /mnt/NAS
ls -ld /var/lib/mpd/music/NAS/
drwxrwxrwx 1 root root 4096 Mar 21  2019 /var/lib/mpd/music/NAS/
ls -l /var/lib/mpd/music/NAS/
total 8
drwxrwxrwx 20 root root 4096 Jan 28 04:48 Bangkok
```

Further, if I follow gael1980's message, I am unable to remove my NAS mount.  At least from the UI if I head to Sources -> Network Sources -> [delete] - it throws a red toast message stating that it can't remove the network share.

I'm not familiar enough with the cmd-line foo to unmount via shell.

If this is something that needs further validation, let me know what commands to run and I can do so.

cbu99, 2020-02-07:
Same for me: GUI gives me the same error message if I try to remove the NAS share. If I add it for a second time I can remove the second entry only.

gael1980, 2020-02-07:
I didn't remove the link from GIU.
I physically removed the cable of the NAS.
Then I restarted volumio.
Then I put the cable back in and restarted volumio.

xipmix, 2020-02-07:
@futuralogic - interesting that it won't unmount from the GUI. It's probably a separate issue, but it would help to understand whether it is or not if you could boot, try to remove via the GUI and send a log of that, shortly after trying and failing to unmount.

The way to do it via the command line (give this a try  after the experiment above)
```
$ mount | grep Bangkok   # shows all the mount options in gory detail
$ sudo umount -v /mnt/NAS/Bangkok
```
That should wither work or give some helpful error messages.

futuralogic, 2020-02-08:
> @futuralogic - interesting that it won't unmount from the GUI. It's probably a separate issue, but it would help to understand whether it is or not if you could boot, try to remove via the GUI and send a log of that, shortly after trying and failing to unmount.

Here is the log after a fresh reboot and trying to unmount via GUI:
http://logs.volumio.org/volumio/LOVRX3r.html

> 
> The way to do it via the command line (give this a try after the experiment above)
> 
>
> `$ mount | grep Bangkok   # shows all the mount options in gory detail`

Result:
```
192.168.1.xxx:/volume1/Music on /mnt/NAS/Bangkok type nfs4 (ro,relatime,vers=4.0,rsize=16384,wsize=131072,namlen=255,soft,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=192.168.1.xxx,local_lock=none,addr=192.168.1.xxx)
```

> `$ sudo umount -v /mnt/NAS/Bangkok`
>

Result:

`umount.nfs4: /mnt/NAS/Bangkok: device is busy`

It's been rebooted for several minutes.

xipmix, 2020-02-09:
Thanks for this. 

I wanted the log as I was wondering whether there was some weird character in the umount path that wasn't being handled correctly (we've had a few issues with that in the past). Not the case here.

As the manual umount command shows, something is keeping the mount 'busy'. Usually this means something has a file or directory open. Most likely this is mpd, so it looks like we have another aspect of the basic problem.

One further question, what are the permissions of
```/var/lib/mpd/music/NAS/Bangkok/#recycle``` ?
I'm still wondering why we see errors in mpd.log and whether they matter.

xipmix, 2020-02-09:
> @xipmix what do you suggest?

At first I thought of this
```
[Unit]
Description=Music Player Daemon
Documentation=man:mpd(1) man:mpd.conf(5)
- After=network.target sound.target
+ After=network.target sound.target remote-fs.target
```
but the remote mounts of interest are done by volumio, not systemd. So we will have to enforce the ordering of app/plugins/music_service/mpd to be after app/plugins/system_controller/networkfs. I have no idea how to achieve that.

There is another option to explore though. I see the service is the 'notify' type.
```
[Service]
Type=notify
ExecStart=/usr/bin/mpd --no-daemon
# allow MPD to use real-time priority 50
LimitRTPRIO=50
LimitRTTIME=infinity
```
There have been complaints about similar-looking issues in the past (MusicPlayerDaemon/MPD/issues/259) and it seems setting ```Type=simple``` was helpful. But why this would be needed now, two years after it was set to notify (https://github.com/volumio/Build/commit/67964fd9022c8053aac5452d14779c5c198e5757#diff-a485c48a3ea8b9c8d3d49bc829516201), escapes me.

futuralogic, 2020-02-09:
> Thanks for this.
>
> I wanted the log as I was wondering whether there was some weird character in the umount path that wasn't being handled correctly (we've had a few issues with that in the past). [...]

Here are perms on #recycle:
`d--------- 2 nobody users 4096 Jan 17 07:00 /var/lib/mpd/music/NAS/Bangkok/#recycle`

I'm fairly certain the #recycle folder is the recycle bin on my Synology NAS managed by the Synology.

gael1980, 2020-02-09:
Off topic. In order for volumio not to index the ".recycle" directory I added a ".mpdignore" file.

`echo "*" > /sharedfolders/Musique/.recycle/.mpdignore`

xipmix, 2020-02-09:
Thanks for the update and the permission check. Not sure what to suggest next.

futuralogic, 2020-02-15:
@xipmix Is there any way to interrogate why MPD is busy? Or to dig into sysctl service start logs? 

I tried parsing through the startup log that I posted above and see mpd "FAILURE", is there any way to dig further?

I'm hoping to have time this weekend where I plan to re-image the sdcard with the latest volumio from scratch.  It may at least help point in the direction whether something was messed up due to/during the upgrade vs a clean install.

volumio, 2020-02-15:
Please try the latest dev release, it should fix the issue. let me know

severindupouy, 2020-02-15:
Well done ! It's working by my side. Thanks a lot.

volumio, 2020-02-15:
Update. Released. Thanks for the feedback

closed 2020-02-15 by volumio

futuralogic, 2020-02-15:
Confirming the update also works on my two volumio's as well. Many thanks!

(Any link to the commit changes? Curious what the issue was.)

xipmix, 2020-02-16:
Possibly 16fbd01d4259beeb280a035d151278aaabc545a1