volumio/Volumio2 #1889: Ethernet breaking down on RPi 3b

suuuehgi, 2020-02-03:
Hi,

since a while my Ethernet is breaking down on Volumio and with it the complete UI sometimes becomes unresponsive.
Sometimes it works for a while, then it happens again.

Yesterday it told me it would have 169.254.250.189 assigned on eth0 what clearly does not work out.

Any ideas? It might be a duplicate of [Issue 1463](https://github.com/volumio/Volumio2/issues/1463).

I already changed the cable, the power adapter and checked the switch.

I don't know how to get the complete log:

```log
[...]
Jan 22 14:47:17 hostname kernel: VFS: Disk quotas dquot_6.6.0
...skipping...
Feb 02 08:54:25 hostname dhcpcd[3714]: eth0: using IPv4LL address 169.254.250.189
Feb 02 08:54:25 hostname avahi-daemon[717]: Joining mDNS multicast group on interface eth0.IPv4 with address 169.254.250.189.
Feb 02 08:54:25 hostname dhcpcd[3714]: eth0: adding host route to 169.254.250.189 via 127.0.0.1
Feb 02 08:54:25 hostname dhcpcd[3714]: eth0: adding route to 169.254.0.0/16
Feb 02 08:54:25 hostname avahi-daemon[717]: New relevant interface eth0.IPv4 for mDNS.
Feb 02 08:54:25 hostname avahi-daemon[717]: Registering new address record for 169.254.250.189 on eth0.IPv4.
[...]
# Comment: Now getting back to default
Feb 02 08:54:29 hostname dhcpcd[3714]: eth0: soliciting a DHCP lease
Feb 02 08:58:49 hostname nmbd[922]: [2020/02/02 08:58:49.088499,  0] ../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Feb 02 08:58:49 hostname nmbd[922]: query_name_response: Multiple (2) responses received for a query on subnet 192.168.178.77 for name WORKGROUP<1
Feb 02 08:58:49 hostname nmbd[922]: This response was from IP 192.168.178.131, reporting an IP address of 192.168.178.131.
Feb 02 09:00:47 hostname systemd[1]: Starting Cleanup of Temporary Directories...
Feb 02 09:00:47 hostname systemd[1]: Started Cleanup of Temporary Directories.
Feb 02 09:08:27 hostname smbd[5857]: [2020/02/02 09:08:27.552686,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Feb 02 09:08:27 hostname smbd[5857]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Feb 02 09:08:27 hostname smbd[5857]: [2020/02/02 09:08:27.563292,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Feb 02 09:08:27 hostname smbd[5857]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Feb 02 09:08:27 hostname smbd[5857]: [2020/02/02 09:08:27.563587,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Feb 02 09:08:27 hostname smbd[5857]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Feb 02 09:08:27 hostname smbd[5857]: [2020/02/02 09:08:27.563840,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Feb 02 09:08:27 hostname smbd[5857]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Feb 02 09:08:27 hostname smbd[5857]: [2020/02/02 09:08:27.690071,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Feb 02 09:08:27 hostname smbd[5857]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Feb 02 09:08:28 hostname smbd[5857]: [2020/02/02 09:08:28.350764,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Feb 02 09:08:28 hostname smbd[5857]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Feb 02 09:16:02 hostname smbd[6821]: [2020/02/02 09:16:02.853139,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Feb 02 09:16:02 hostname smbd[6821]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Feb 02 09:16:02 hostname smbd[6821]: [2020/02/02 09:16:02.857358,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Feb 02 09:16:02 hostname smbd[6821]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Feb 02 09:16:02 hostname smbd[6821]: [2020/02/02 09:16:02.857687,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Feb 02 09:16:02 hostname smbd[6821]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Feb 02 09:16:02 hostname smbd[6821]: [2020/02/02 09:16:02.857967,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Feb 02 09:16:02 hostname smbd[6821]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Feb 02 09:16:02 hostname smbd[6821]: [2020/02/02 09:16:02.867358,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Feb 02 09:16:02 hostname smbd[6821]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Feb 02 09:16:02 hostname smbd[6821]: [2020/02/02 09:16:02.876561,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Feb 02 09:16:02 hostname smbd[6821]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Feb 02 09:40:20 hostname dhcpcd[3714]: eth0: offered 192.168.178.76 from 192.168.178.1
Feb 02 09:40:25 hostname dhcpcd[3714]: eth0: leased 192.168.178.76 for 864000 seconds
Feb 02 09:40:25 hostname avahi-daemon[717]: Registering new address record for 192.168.178.76 on eth0.IPv4.
Feb 02 09:40:25 hostname dhcpcd[3714]: eth0: adding host route to 192.168.178.76 via 127.0.0.1
Feb 02 09:40:25 hostname avahi-daemon[717]: Withdrawing address record for 169.254.250.189 on eth0.
Feb 02 09:40:25 hostname dhcpcd[3714]: eth0: adding route to 192.168.178.0/24
Feb 02 09:40:25 hostname avahi-daemon[717]: Leaving mDNS multicast group on interface eth0.IPv4 with address 169.254.250.189.
Feb 02 09:40:25 hostname avahi-daemon[717]: Joining mDNS multicast group on interface eth0.IPv4 with address 192.168.178.76.
Feb 02 09:40:25 hostname dhcpcd[3714]: eth0: adding default route via 192.168.178.1
Feb 02 09:40:25 hostname dhcpcd[3714]: eth0: deleting host route to 169.254.250.189 via 127.0.0.1
Feb 02 09:40:25 hostname dhcpcd[3714]: eth0: deleting route to 169.254.0.0/16
[...]
```

Thanks a lot!

xipmix, 2020-02-04:
If you can't get to http://volumio.local/dev and get that to work
```
$ volumio logdump issue1889
```
should do it. This will dump to a file that you could upload here.

Good debugging work so far. Does the new power supply really give 2amps, that seems to be necessary.

Looks like your DHCP lease time is long enough (86,400), so it's not a timeout.

Are you able to check your router's DHCP log, to see what it thinks is happening?

Are any other devices on your network having trouble like this?

Could there be another device with the same IP address? That's always fun.

If the DHCP pool is too small, something else may be getting 192.168.178.76 assigned to it.

I find it useful to set a specific address in the router for the volumio device so it gets the same IP every time. Is that possible for you - perhaps you already did that.

suuuehgi, 2020-02-04:
Thank you for the feedback! I'm currently at work but I'll try to answer a few:

> Good debugging work so far. Does the new power supply really give 2amps, that seems to be necessary.

Thank you! At least it used to give 2 A and it supports QC Quick Charge 3.0. I connected my phone to both of them earlier today and to my surprise got not more than 1 A on each of them. I can't rule out my phones fault. I used to get ~ 1.8 A on both.

> Are any other devices on your network having trouble like this?

Not that I know of. But there are many inexperienced users within the network. I can't really answer this one.

> Could there be another device with the same IP address? That's always fun. [...] I find it useful to set a specific address in the router for the volumio device so it gets the same IP every time. Is that possible for you - perhaps you already did that.

Yes, I set it to fixed. :)

suuuehgi, 2020-02-04:
By the way ... Do you know anything about the RPis power rail? Is it possible to feed in power via the USB ports? In the hypothetical case the power adapter meanwhile really just makes 1 A a port, I'm thinking about doubling that. :)

xipmix, 2020-02-05:
Sorry, no idea about the power rail. Have you asked in the Pi forums about others having this issue?

suuuehgi, 2020-02-26:
> Sorry, no idea about the power rail. Have you asked in the Pi forums about others having this issue?

No. It's just one of many things wanting attention.

I noticed that the PI keeps running after pulling the power cord when I also have the aforementioned USB Y-cable attached. Hence the rails somehow go together.
I have not had any problems again since I added this second "power cord".

Not really satisfying but at least working. Let's wait a bit longer ...

suuuehgi, 2020-04-01:
No more issues so far using both ports of the power adapter. :-)

I conclude: One can support the Pi by injecting power via the USB ports with a USB male-male cable or a USB Y-cable in my case.

closed 2020-04-01 by suuuehgi