volumio/Volumio2 #1892: Enqueue Song After Currently Playing Song

quizilkend, 2020-02-09:
Hey there,
thank you for this beautiful piece of software.

I have a feature suggestion:
It would be awesome to have an option in the dropdown menu of a track/album/artist: "Enqueue After Currently Playing Song".

xipmix, 2020-02-10:
Can I clarify this - do you mean
 * I have a queue (of say 5 songs), insert this song so it will be played next?
 * or something else?

What would you like to happen if the queue is being played in random order? Still insert the song just after the first one in the queued list?

quizilkend, 2020-02-13:
Hey, 
thanks for your reply.
1. yes exactly
2. even in shuffle mode, the song should be played next.

The music player clementine has that built in.

[image]
Normally the tracks are played from top to bottom (or shuffled), but the numbers indicate the songs that are going to be played next.