volumio/Volumio2 #1893: ID3 tags not always recognized

MiGo2000, 2020-02-10:
Using volumio on a raspberry pi 3 successfully for some days, I find many files tagged with the id3 numbers instead their names, e.g. "(4)" instead of "Disco". Retagging didn't help. I tried v2.3 and v2.4, nothing helped. Additionally I find correctly named files as well. So my genre section starts with a lot of numbers, some in brackets, some not, followed by the correct tags.

xipmix, 2020-02-12:
It might be helpful if you could use some other tool to dump the tags for one of the files that shows numbers in the listing above. What format are the problematic files - mp3, flac, other?

Some tools that may help - metaflac (in flac package), id3v2

MiGo2000, 2020-02-15:
Yes, I should try that too.
I used easytag for tagging my files for a long time now, and it's very strange that this program shows so different behaviour amongst my music files.
I will also try a different one.