volumio/Volumio2 #1903: Airplay state not chcnged to 'stop' after source disconnected

monte-monte, 2020-02-26:
Volumio version 2.712 (seen the same behavior in previous versions)

Problem: playing music from iPad via Airplay > switch back from Volumio to iPad or any other airplay device > Volumio's status remains 'play', seek state keeps counting into infinity. The only way is to start playing anything from playlist, or stop playing on iPad before disconnecting.

Expected behavior: Volumio detects airplay client disconnected and changes status to 'stop'.

monte-monte, 2020-02-26:
Tried to set status to stop with ``volumio stop``. Response was ``"response":"stop Success"`` but player state remained 'play' and even 'seek' time didn't reset.

monte-monte, 2020-02-26:
Okay, so Volumio ignores ``"ssnc" "pend"`` call that shairport-sync sends to metadata pipe, when client disconnects but it doesn't ignore it when client sends pause. Maybe this is because no metadata is sent before ``"ssnc" "pend"``, when client disconnects abruptly. If this is the case, I think we need to change how Volumio reads shairport's pipe.

acresp, 2021-01-25:
The issue is still there in the last version 🙁 
When an airplay client disconnect without pause, the volumio state stays on "play".

"status":"play"

And a cmd stop through the API doesn't change it to stop either 🙁

elektro-wolle, 2021-01-25:
@acresp The PR is still open... But you can add the line manually: see PR #2005