volumio/Volumio2 #1904: Support for redshift

suuuehgi, 2020-02-26:
Hi,

I would like to bring in the idea of redshift on the touch ui.
I would love it since at night my little hdmi touchscreen is really hard on my eyes.

Thanks a lot!

volumio, 2020-02-26:
Can you elaborate more on what is redshift?

suuuehgi, 2020-02-27:
> Can you elaborate more on what is redshift?

Yes, certainly. Sorry, in my filter bubble it's as mandatory as an adblocker.

[Redshift](https://en.wikipedia.org/wiki/Redshift_(software)) filters out blue light what messes with your circadian rhythm (e.g. bad sleep; by significantly suppressing [melatonin](https://en.wikipedia.org/wiki/Melatonin) secretion) and _massively_ reduces eye-strain at night.

[image]

It gradually dims according to your location, time and max values. In my experience people regularly don't notice it until they disable it during the night and their eyes burn away.

I head it would work with the experimental OpenGL driver from `raspi-config`.

volumio, 2020-03-12:
Not really a volumio thing, since browsers or apps can achieve that

closed 2020-03-12 by volumio

suuuehgi, 2020-03-20:
I'm not aware that browsers can do that and I don't have apps in Volumio. Maybe you over-read that I'm talking about my hdmi touchscreen driven by the Volumio Touch Display Plugin.

volumio, 2020-03-20:
Ah, now I understand. 
I don't think we'll be able to do it anyway...