volumio/Volumio2 #1905: Issue for Browse source

vrenaville, 2020-02-26:
Since Last update on my rasperry PI 3A, I have issue to browse source.

https://github.com/volumio/Volumio2/commit/e009044b21e26008541d101de21df65f6ed671d5

It seems to come from this commit, If I revert to previous value it works again.

volumio, 2020-02-26:
Thanks for pointing it out, apparently I've made a stupid mistake.
Can you update to latest commit and report if that helps?
https://github.com/volumio/Volumio2/commit/15b2086123dd970f45064ed3f1c8e98bf53d7f51

vrenaville, 2020-02-28:
@volumio 

I have tested and it's work, thanks for your work

vrenaville, 2020-03-02:
@volumio 

I have made more test this week-end, And the isssue Is still there, If I revert to the original configuration
it works again.

xipmix, 2020-03-02:
Could you [send a log](https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html) when you notice it failing?

vrenaville, 2020-03-05:
@xipmix @volumio  I close it, I have just seen that you have revert all config modification on new release, So it's worked

closed 2020-03-05 by vrenaville