volumio/Volumio2 #1908: Spotify Premium Indication

suuuehgi, 2020-03-05:
Good morning,

I just tried the Spotify plugin for a while without success before I found this warning in the logs:
```[WARN] Login failed: User needs a premium account```

Maybe you could incorporate this information in the UI. It currently states `Spotify Login failed: Please check username and password` what is clearly not the case.
And I also recommend to change the plugin description `Spotify plugin for Volumio2` to something indicating the sole support for premium accounts.

Thanks for the work!

xipmix, 2020-03-10:
I think this issue really belongs in https://github.com/volumio/volumio-plugins, could you open it there or at least cross-ref to this issue?

ashthespy, 2020-03-10:
FWIW it does mention it on the plugin details page when you go to install it.. 
[image]

closed 2020-03-12 by volumio