volumio/Volumio2 #1909: Rest API play/pause and Web Radio stuck in playing

Slawo60, 2020-03-05:
Hi,

I have an issue with a REST API. 
I have a script that pauses volumio for a call and resume last state when done. 
It works well when playing a music file, but it stuck volumio when playing web radio
I do
`curl volumio.local/api/v1/commands/?cmd=pause`
then
`curl volumio.local/api/v1/commands/?cmd=play`

It works one time, then it's stuck in pause state, but the player is actually playing. The REST API and the web GUI cannot pause the web radio.To get unstuck, I must change radio.

#558  seems to be the same issue.

xipmix, 2020-03-07:
I tried to reproduce this and found something strange. This is on v2.714.

 * play a track from my NAS source
 * ```curl volumio.local/api/v1/commands/?cmd=pause``` - pauses
 * ```curl volumio.local/api/v1/commands/?cmd=play``` - play

So far so good. Now try a web radio.

 * play a web radio (Audiophile Baroque, http://94.23.201.38:8000/stream)
 * ```curl volumio.local/api/v1/commands/?cmd=pause``` - pauses
 * ```curl volumio.local/api/v1/commands/?cmd=play``` - play
 * press play/pause on the web ui. Nothing happens, it keeps playing.
 * ```curl volumio.local/api/v1/commands/?cmd=pause``` - keeps playing

So it seems like it works exactly once, then gets into a state where pause is ignored.
The log below was produced by:
 * reboot
 * clear queue
 * play a web radio  (Audiophile Baroque)
 * ```curl volumio.local/api/v1/commands/?cmd=pause``` - pauses
 * ```curl volumio.local/api/v1/commands/?cmd=play``` - play
 * press play/pause on the web ui. Nothing happens, it keeps playing. The UI icon showing was the 'play' icon, it should have been the 'pause' icon.
 * send log.

Log http://logs.volumio.org/volumio/ki3YydI.html

xipmix, 2020-03-07:
Looking at the log I think the issue might be related to this, but that's just a guess.
```
Mar 07 01:21:44 volumio volumio[849]: info: Received update from a service different from the one supposed to be playing music. Skipping notification.Current webradio Received mpd
```
The sequence to get there is
```
Mar 07 01:21:40 volumio volumio[849]: ---------------------------- Client requests Volumio pause
Mar 07 01:21:40 volumio volumio[849]: info: CoreCommandRouter::volumioPause
Mar 07 01:21:40 volumio volumio[849]: info: CoreStateMachine::pause
Mar 07 01:21:40 volumio volumio[849]: info: CoreStateMachine::stPlaybackTimer
Mar 07 01:21:40 volumio volumio[849]: info: CoreStateMachine::servicePause
Mar 07 01:21:40 volumio volumio[849]: info: CorePlayQueue::getTrack 0
Mar 07 01:21:40 volumio volumio[849]: info: CoreCommandRouter::servicePause
Mar 07 01:21:40 volumio volumio[849]: info: [1583544100954] ControllerWebradio::pause
Mar 07 01:21:40 volumio volumio[849]: info: ControllerMpd::sendMpdCommand pause
...
Mar 07 01:21:41 volumio volumio[849]: info: STATE SERVICE {"status":"pause","position":0,"seek":7990,"duration":0,"samplerate":"44.1 kHz","bitdepth":"24 bit","channels":2,"random":false,"updatedb":false,"repeat":false,"isStreaming":false,"title":"Johann Gottlieb Goldberg - Trio Sonata In C Major, [London Baroque]","artist":"Audiophile Baroque","album":null,"uri":"http://94.23.201.38:8000/stream","trackType":""}
...
Mar 07 01:21:44 volumio volumio[849]: ---------------------------- Client requests Volumio play
...
Mar 07 01:21:44 volumio volumio[849]: ---------------------------- MPD announces state update: player
...
Mar 07 01:21:44 volumio volumio[849]: info: Received update from a service different from the one supposed to be playing music. Skipping notification.Current webradio Received mpd
...
(there is no info: STATE SERVICE line)
```
After that we're stuck
```
Mar 07 01:21:50 volumio volumio[849]: ---------------------------- Client requests Volumio pause
Mar 07 01:21:50 volumio volumio[849]: info: CoreCommandRouter::volumioPause
Mar 07 01:21:50 volumio volumio[849]: info: CoreStateMachine::pause

<nothing logged until web UI press of 'play/pause' icon>

Mar 07 01:21:58 volumio volumio[849]: info: CoreCommandRouter::volumioPlay
Mar 07 01:21:58 volumio volumio[849]: UNSET VOLATILE
Mar 07 01:21:58 volumio volumio[849]: info: CoreStateMachine::play index undefined
Mar 07 01:21:58 volumio volumio[849]: info: CoreStateMachine::setConsumeUpdateService undefined
Mar 07 01:21:58 volumio volumio[849]: info: CorePlayQueue::getTrack 0
Mar 07 01:21:58 volumio volumio[849]: info: CoreStateMachine::startPlaybackTimer
Mar 07 01:21:58 volumio volumio[849]: info: CorePlayQueue::getTrack 0
Mar 07 01:21:58 volumio volumio[849]: info: [1583544118711] ControllerWebradio::resume
Mar 07 01:21:58 volumio volumio[849]: info: ControllerMpd::sendMpdCommand play
Mar 07 01:21:58 volumio volumio[849]: info: sending command...
Mar 07 01:21:58 volumio volumio[849]: info: parsing response...
...

Slawo60, 2020-03-09:
I found the solution. Because the web radio is a audio stream, "pause" doesn't really work. I've used `volumio.local/api/v1/commands/?cmd=stop` and it's working just fine. 

Thank you very much for you help.

xipmix, 2020-03-10:
Nonetheless it would be good if the web UI play/pause worked as expected...

camelcamro, 2023-07-31:
yes, so, that is a bug.
because i can not use as a solution the "STOP", 
eg: a stop would change the position (eg: playing a mp3 file), and i would loose the position.

so, this bug is still open since 3 YEARS.
is developing ongoing on new features which can be sold only ? or also trying to fix first the basics ?

Slawo60, 2023-07-31:
The workaround i used for my script is to check if the type media is a webra

closed 2023-07-31 by Slawo60