volumio/Volumio2 #1913: Album restart if go froward (CUE file)

a491765, 2020-03-06:
Hi,

I have a problem when i want to move in a song using the progress bar.
When i click to go forward, the album restart from the beginning (but the current song still show in Volumio playback UI)

Particularity : My whole collection is composed of single flac album files with cue sheets.

How to Reproduce : Go to Music Library, select attached USB, select album, launch a .cue.
All musics are added to queue. swith for a song (exemple track 04). Now click in the progress bar to go forward. track n°4 still shown but the player restart album from 0 (track is playing 1)

System info :
    Version of Volumio: 2.714
    Hostname: volumio
    Kernel: 4.19.86-v7l+
    Governor: performance

xipmix, 2020-03-07:
Interesting. I tried to reproduce with a more conventional setup (v2.703):
 * clear queue
 * add album to queue (one .flac per track)
 * select track 4, it starts playing
 * click on progress bar
 * track 4 continues
extra test
 * click on progress bar to go _back_ in the track
 * track 4 plays from newly selected position, as expected

So it would seem the issue is the .cue files.
Can you [send a log](https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html) of this happening?
 * boot
 * test as above
 * send log at that point

a491765, 2020-03-07:
Ok, i send the log.

You're right it deals with the cue file.
I try to play the full flac directly (without using the cue) and go forward using progress bar and it works fine as expected.