volumio/Volumio2 #1921: Permission denied when accessing USB via network

amespuga, 2020-03-26:
This software is amazing, thank you all for your contributions!
I'm having this issue, which may be my fault. The fact is I'm told **I don't have permissions to access a USB plugged to the RPI4**.
[image]

Under /media, I see the following:
[image]
As you can see, I tried to give full permissions, but it stays the same. I cannot change the ownership either (it's root), even if I sudo it.

What is weird is that my unit is called PEN_DRIVE, and when I enter the USB from my PC via network, I see kind of a level above, where I can see PEN_DRIVE, EFI and I can copy files (Buika). This blew my mind. Check the first figure. Is like I'm copying data outside the partition.

Any of you experiencing the same problem? Thank you very much!

xipmix, 2020-03-26:
I assume the pen drive is shared via samba? What does ```sudo smbstatus``` show?
(PS you can use three backticks to show code chunks, so you don't need to screenshot)

xipmix, 2020-03-28:
I did a test of this today, plugged in a random USB stick. I was able to access it via smbclient, from the directory level above the USB stick. So this should work. (my volumio is rpi2, 2.714, btw).

The reason for ending up in the level above is as follows:

 * samba is exporting three areas - this is before any USB stick is mounted
```
<excerpt from /etc/samba/smb.conf>
[Internal Storage]
        comment = Volumio Internal Music Folder
        path = /data/INTERNAL
        read only = no
        guest ok = yes

[USB]
        comment = Volumio Internal Music Folder
        path = /mnt/USB
        read only = no
        guest ok = yes

[NAS]
        comment = Volumio Internal Music Folder
        path = /mnt/NAS
        read only = no
        guest ok = yes
```
 * to smbclient that looks like this
```
$ smbclient -L //192.168.x.x -U volumio
Enter volumio's password: 
Domain=[WORKGROUP] OS=[Windows 6.1] Server=[Samba 4.2.14-Debian]

        Sharename       Type      Comment
        ---------       ----      -------
        Internal Storage Disk      Volumio Internal Music Folder
        USB             Disk      Volumio Internal Music Folder
        NAS             Disk      Volumio Internal Music Folder
        IPC$            IPC       IPC Service (Volumio Audiophile Music Player)
Domain=[WORKGROUP] OS=[Windows 6.1] Server=[Samba 4.2.14-Debian]

        Server               Comment
        ---------            -------
        VOLUMIO              Volumio Audiophile Music Player
...
```
Now there's something to note about the USB share
```
$ ls -l /mnt/USB
lrwxrwxrwx 1 root root    6 Mar  2 22:44 /mnt/USB -> /media
```
So when you access the 'USB' share you end up looking at the /media directory on the filesystem.

When you plug in your USB drive, it gets mounted at /media/PEN_DRIVE, ie a directory below /media.

As to your access problem.

Whether you are able to access a share or not is controlled by a few different things about your setup.
 * the smb.conf (see above). If it differs from the above you may have problems. You might have to authenticate before you can access it, see below.
 * what kind of filesystem is on the drive. If it's something not supported by the linux system inside volumio, you won't be able to write to it and may not be able to get read access either.
 * name resolution. I think that's likely the problem here.

Your volumio box is probably advertising itself as with netbios as 'VOLUMIO' . You should check if that name resolves to an IP address and whether that matches the IP of your volumio device, for example:
```
$ host volumio
host volumio has address 192.168.2.21
$ /sbin/ifconfig |grep 192.168.2.21
inet addr:192.168.2.21 Bcast:192.168.2.255  Mask:255.255.255.0
```
If things don't match then you will have problems.

You may have better luck connecting via IP address. For example I was able to connect to volumio's shares and create a directory on the USB stick:
```
$ nmblookup -S VOLUMIO
192.168.2.21 VOLUMIO<00>            # does this match the result from 'host', ie DNS, above?
Looking up status of 192.168.2.21
        VOLUMIO         <00> -         H <ACTIVE> 
        VOLUMIO         <03> -         H <ACTIVE> 
        VOLUMIO         <20> -         H <ACTIVE> 
        WORKGROUP       <00> - <GROUP> H <ACTIVE> 
        WORKGROUP       <1e> - <GROUP> H <ACTIVE> 

        MAC Address = 00-00-00-00-00-00

$ smbclient -m smb3 //192.168.2.21/USB -U volumio
Enter volumio's password: 
smb: \> dir
  .                                   D        0  Sat Mar 28 16:07:37 2020
  ..                                  D        0  Thu Jan  1 10:01:08 1970
  2344-CDB9                           D        0  Thu Jan  1 10:00:00 1970  # this is the USB stick

                12791212 blocks of size 1024. 12105320 blocks available
smb: \> cd 2344-CDB9\
smb: \2344-CDB9\> dir
  .                                   D        0  Thu Jan  1 10:00:00 1970
  ..                                  D        0  Sat Mar 28 16:07:37 2020
  T7810A31.exe                        A  9922880  Thu Aug  1 12:46:06 2019

                58234 blocks of size 1024. 48542 blocks available
smb: \2344-CDB9\> mkdir t
smb: \2344-CDB9\> dir
  .                                   D        0  Sat Mar 28 16:08:45 2020
  ..                                  D        0  Sat Mar 28 16:07:37 2020
  T7810A31.exe                        A  9922880  Thu Aug  1 12:46:06 2019
  t                                   D        0  Sat Mar 28 16:08:45 2020

                58234 blocks of size 1024. 48540 blocks available
smb: \2344-CDB9\> rmdir t
smb: \2344-CDB9\> exit
$
```
When I deliberately break the name resolution, ie boot volumio with a different IP address set up, I can't connect via the name, but the IP address works fine.
```
$ smbclient -m smb3 //volumio/USB -U volumio
Enter volumio's password: 
Connection to VOLUMIO failed (Error NT_STATUS_HOST_UNREACHABLE)
```

Does any of this help?

theincompletecode, 2020-04-19:
Hi @xipmix! Thank you so much for the answer, I wasn't recieving the notifications.
OK, so step by step, I am a little bit slow here:

- My volumio is working on a RPI4, version 2.729 (Released Wednesday March 18)

- The /etc/samba/smb.conf file looks the same.

- The smbstatus looks a bit empty, though:

`volumio@volumio:/etc/samba$ sudo smbstatus
[sudo] password for volumio: 

Samba version 4.2.14-Debian
PID     Username      Group         Machine            Protocol Version       
------------------------------------------------------------------------------

Service      pid     machine       Connected at
-------------------------------------------------------

No locked files` 

- The drive is formatted so Volumio can understand it, I can see the albums and artists once the PEN_DRIVE is plugged :) It is mounted at /media, as well:

`volumio@volumio:/media$  ls -l /mnt/USB
lrwxrwxrwx 1 root root 6 Mar 18 00:32 /mnt/USB -> /media`

- Now, as you were pointing out, it looks that the name "Volumio" is not resolved:
`volumio@volumio:/etc/samba$ host volumio
Host volumio not found: 3(NXDOMAIN)
volumio@volumio:/etc/samba$ host Volumio
Host Volumio not found: 3(NXDOMAIN)`

I guess the problem is here. Should I change the hostname?

Again, thank you very much!

xipmix, 2020-04-20:
I now think the smbstatus thing is a red herring but thanks for checking; it shows there's nothing holding a lock some place.

It's a bit strange you aren't getting the name resolved.
The way this is controlled involves a few files. The first one to check is the Name Service Switch file, nsswitch.conf. This is the relevant line from mine.

```
volumio$ grep host /etc/nsswitch.conf 
hosts:          files dns mdns4_minimal [NOTFOUND=return] wins
```
The order is significant, it tells the system what order to try each name resolution system. If a method fails, it tries the next, working from left to right.
 * 'files' means the /etc/hosts file. You can probably fix your issue with the right entry in that file.
 * 'dns' means the DNS servers which are normally listed in /etc/resolv.conf. To resolve 'volumio' you will need a DNS server that knows about your home network, usually this is your router. A bit of tweaking on your router might be a better fix, but let's not get into that right away.

It would help to see what /etc/hosts contains and what result the command
```
volumio$ hostname
```
returns. I am expecting 'volumio'.

 * mdns4_minimal is 'multicast DNS', which is a bit of a mystery to me.

 * 'wins'  is a windows name resolution mechanism that may matter for samba.
The equivalent of 'host' is 'nmblookup', it would be interesting to see if you get. a hit from it.
```
volumio$ nmblookup -B -R volumio. # Broadcast the question "Tell me about 'volumio'"
192.168.2.21 volumio<00>
```

Note that samba defaults to a different set of name lookup methods:
  "lmhosts", "host", "wins" and "bcast"
 I think we can ignore lmhosts, it expects a special 'lmhosts' file to exist in the samba installation. 'host' means use the methods in nsswitch.conf (/etc/hosts file or DNS). 'wins' we have met already. Not completely sure what 'bcast' does but I think it is similar to the `nmblookup -B` shown above. If the default order has been changed for some reason that could create issues. If it had been modified there should be a line configuring the 'name resolve order' in smb.conf.

theincompletecode, 2020-04-21:
Hi @xipmix. Thank you again for your effort.
Reading once again the first answer you provided, I did the following thing:

**STEP1**
I edited the smb.conf file so it accesses the PEN_DRIVE straight ahead, without offering the level above to the network. **Check the USB path.** The file goes like this:
```
<excerpt from /etc/samba/smb.conf>
[Internal Storage]
        comment = Volumio Internal Music Folder
        path = /data/INTERNAL
        read only = no
        guest ok = yes

[USB]
        comment = Volumio Internal Music Folder
        path = /mnt/VOLUMIO_DISC
        read only = no
        guest ok = yes

[NAS]
        comment = Volumio Internal Music Folder
        path = /mnt/NAS
        read only = no
        guest ok = yes
```

**STEP2**
I created a soft link at /mnt, as follows:
```
volumio@volumio:~$ cd /mnt/
volumio@volumio:/mnt$ ll
total 24
drwxrwxrwx 1 root    root    4096 Apr 20 16:09 .
drwxrwxrwx 1 root    root    4096 Jan  1  1970 ..
lrwxrwxrwx 1 root    root      14 Mar 18 00:32 INTERNAL -> /data/INTERNAL
drwxrwxrwx 1 root    root    4096 Mar 21 11:13 NAS
drwxrwxrwx 2 root    root       3 Mar 18 00:32 UPNP
lrwxrwxrwx 1 root    root       6 Mar 18 00:32 USB -> /media
lrwxrwxrwx 1 volumio volumio   12 Apr 20 16:09 VOLUMIO_DISC -> /media/PEN_DRIVE
```

**STEP3**
I granted full permissions (777) to the /media/PEN_DRIVE disk with chmod.
For doing that, I had to format the disk to ntfs, since exFAT or FAT32 do not support permissions managing. I know 777 permissions are dangerous, but inside my local network its only me.

Now I see the VOLUMIO_DISC when connecting to SMB->volumio, with all its contents and I can read and write them.

Pheeeeew! It took a while. Thank you for your posts, you enlightened the answer!

closed 2020-04-21 by theincompletecode