volumio/Volumio2 #1922: Disable samba

suuuehgi, 2020-04-01:
Hi,

I noticed the samba service spamming my log:

```log
Mar 30 15:14:39 hostname smbd[13096]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Mar 30 15:14:39 hostname smbd[13096]: [2020/03/30 15:14:39.899897,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Mar 30 15:14:39 hostname smbd[13096]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Mar 30 15:14:39 hostname smbd[13096]: [2020/03/30 15:14:39.901738,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Mar 30 15:14:39 hostname smbd[13096]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Mar 30 15:14:39 hostname smbd[13096]: [2020/03/30 15:14:39.915176,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Mar 30 15:14:39 hostname smbd[13096]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Mar 30 15:14:39 hostname smbd[13096]: [2020/03/30 15:14:39.925183,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Mar 30 15:14:39 hostname smbd[13096]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Mar 30 16:51:49 hostname nmbd[1011]: [2020/03/30 16:51:49.660770,  0] ../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mar 30 16:51:49 hostname nmbd[1011]: query_name_response: Multiple (2) responses received for a query on subnet 192.168.XXX.XXX for name WORKGROUP<
Mar 30 16:51:49 hostname nmbd[1011]: This response was from IP 192.168.XXX.XXX, reporting an IP address of 192.168.XXX.XXX.
Mar 30 17:01:55 hostname nmbd[1011]: [2020/03/30 17:01:55.133731,  0] ../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mar 30 17:01:55 hostname nmbd[1011]: query_name_response: Multiple (2) responses received for a query on subnet 192.168.XXX.XXX for name WORKGROUP<
Mar 30 17:01:55 hostname nmbd[1011]: This response was from IP 192.168.XXX.XXX, reporting an IP address of 192.168.XXX.XXX.
Mar 30 21:30:16 hostname nmbd[1011]: [2020/03/30 21:30:16.054895,  0] ../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mar 30 21:30:16 hostname nmbd[1011]: query_name_response: Multiple (2) responses received for a query on subnet 192.168.XXX.XXX for name WORKGROUP<
Mar 30 21:30:16 hostname nmbd[1011]: This response was from IP 192.168.XXX.XXX, reporting an IP address of 192.168.XXX.XXX.
```

I am not aware that I use samba, so I don't want to touch the problems apparent here but just disable it.

How would you recommend it to do?

I don't see a way via the WebUI (I can't find it under Settings -> Sources -> Functionalities Control).

Is `systemctl disable smbd` and  `systemctl disable nmbd` save?

Thanks a lot! :)

volumio, 2020-04-02:
Disabling samba will have lots of side effects:
- Discovery of NAS won't work
- Network copy of files won't work
- Use of .local won't work

I don't suggest to do it, those logs are harmless

closed 2020-04-02 by volumio

suuuehgi, 2020-04-02:
All right, thank you!