volumio/Volumio2 #1923: Update mpd

suuuehgi, 2020-04-01:
Hi,

I noticed that volumio uses a really old (v0.20.18, Feb 24, 2018) version of mpd. This causes e.g. webm files, as e.g. Youtube uses them, to be non-seekable.

Concrete scenarios:

  - I played until somewhen, try to seek and it falls back to the start of the file.
  - I have a long file and can't jump.

Afaics this has been [fixed on Feb. 4](https://github.com/MusicPlayerDaemon/MPD/issues/719), release v0.21.20 and above.

Is there any chance this is to be updated soon?

Thanks a lot! :-)

speeedturtle, 2020-04-06:
> Hi,
>
> I noticed that volumio uses a really old (v0.20.18; Feb 24, 2018) version of mpd. This causes e.g. webm files, as e.g. Youtube uses them, to be non-seekable.
>
> Concrete scenarios:
>
> * I played [...]

is it possible to update MPD to latest version downloaded from MPD official site?

volumio, 2020-04-06:
MPD will be updated when we finish the buster porting of Volumio, as of now it's not possible since we miss quite a lot of dependencies

closed 2020-04-06 by volumio