volumio/Volumio2 #1924: Unable to connect to Volumio WIFI with default password

ThalusUlysses, 2020-04-05:
Installed latest Version of Volumio (2.729) using Etcher. Plugged into my RasberryPi 3. tried to connect to Hotspot using the default password. Can not connect to Volumio Hotspot.

ThalusUlysses, 2020-04-05:
Tried it on a complete new setup (new hardware) is not working

rmcaninch, 2020-04-09:
Same issue on a RaspberryPi 2.

deennoo, 2020-04-19:
Same here on a PIZero + USB dongle

thub2707, 2020-04-25:
New installation on Raspberry Pi2 with Edimax EW-7811 Wifi Dongle: Not able to connect Hotspot. After manual wifi setup via LAN Cable, Wifi was working somehow, but very instable. After reboot connection is only randomly possible. Acess via Eth0 is only possible if Wifi dongle is disconnected.

For a test I took an old image (2.xx from 2018) and it works without any issues.
(older-volumio-versions-img-files-t7009.html)