volumio/Volumio2 #1937: Discrepancy between `play` from WebUI and RestAPI/Websocket

ashthespy, 2020-04-20:
Currently the WebUI tracks the currently playing service, and decides to call either `play` or `volatilePlay` from [`player.service.js`](https://github.com/volumio/Volumio2-UI/blob/53dec5e4a67006377c748c5a7ef44206c3c61a02/src/app/services/player.service.js#L46-L54) and [`player-buttons.html`](https://github.com/volumio/Volumio2-UI/blob/5dadb96add827e11380a48cfb8df55adb0cc5b7a/src/app/themes/volumio3/components/player-buttons/volumio3-player-buttons.html#L48-L49).

This breaks both the RestAPI and Websocket API's, as the user shouldn't be expected to keep track of the service, but should instead be incorporated into the statemachine. 

cc https://github.com/balbuze/volumio-plugins/issues/263

volumio, 2020-04-21:
You're absolutely right. Can you come up with a PR to address that?

ashthespy, 2022-01-22:
https://github.com/volumio/volumio3-backend/commit/3b556412844aaf36168e81a4365be5b4a28572d6

closed 2022-01-22 by ashthespy