volumio/Volumio2 #1954: webradio - problem streaming from icecast server to volumio

erols, 2020-05-03:
I am using a raspberry pi to stream audio in my home network according to instructions here: https://www.hifiberry.com/docs/projects/create-your-own-radio-station-with-the-hifiberry-dac-adc-and-icecast/

I can hear the stream if I use a browser, but cannot hear anything if I set it up as a webradio. I see 0 bytes in the Volumio play bar.

I see this in the logs:
```
2020-05-03T15:59:30.430Z - info: CoreCommandRouter::volumioGetState
2020-05-03T15:59:30.431Z - info: CoreStateMachine::getState
2020-05-03T15:59:30.433Z - info: CoreCommandRouter::volumioToggle
2020-05-03T15:59:30.433Z - info: CoreStateMachine::getState
2020-05-03T15:59:30.434Z - info: CoreStateMachine::stop
2020-05-03T15:59:30.434Z - info: CoreStateMachine::setConsumeUpdateService undefined
2020-05-03T16:01:16.474Z - info: CoreCommandRouter::volumioToggle
2020-05-03T16:01:16.475Z - info: CoreStateMachine::getState
2020-05-03T16:01:16.476Z - info: CorePlayQueue::getTrack 7
2020-05-03T16:01:16.476Z - info: CoreStateMachine::play index undefined
2020-05-03T16:01:16.476Z - info: CoreStateMachine::setConsumeUpdateService undefined
2020-05-03T16:01:16.477Z - info: CorePlayQueue::getTrack 7
2020-05-03T16:01:16.478Z - info: CoreStateMachine::startPlaybackTimer
2020-05-03T16:01:16.479Z - info: CorePlayQueue::getTrack 7
2020-05-03T16:01:16.479Z - info: [1588521676479] ControllerWebradio::clearAddPlayTrack
2020-05-03T16:01:16.480Z - info: ControllerMpd::sendMpdCommand stop
2020-05-03T16:01:16.484Z - info: sending command...
2020-05-03T16:01:16.489Z - info: 
---------------------------- MPD announces state update: player
2020-05-03T16:01:16.491Z - info: parsing response...
2020-05-03T16:01:16.491Z - info: ControllerMpd::getState
2020-05-03T16:01:16.492Z - info: ControllerMpd::sendMpdCommand status
2020-05-03T16:01:16.493Z - info: ControllerMpd::sendMpdCommand clear
2020-05-03T16:01:16.493Z - info: sending command...
2020-05-03T16:01:16.494Z - info: sending command...
2020-05-03T16:01:16.496Z - info: 
---------------------------- MPD announces system playlist update
2020-05-03T16:01:16.497Z - info: Ignoring MPD Status Update
2020-05-03T16:01:16.498Z - info: parsing response...
2020-05-03T16:01:16.499Z - info: parsing response...
2020-05-03T16:01:16.500Z - info: ControllerMpd::parseState
2020-05-03T16:01:16.500Z - info: ControllerMpd::sendMpdCommand playlistinfo
2020-05-03T16:01:16.501Z - info: ControllerMpd::sendMpdCommand load "http://192.168.xxx.xxx:8000/hifiberry.mp3"
2020-05-03T16:01:16.502Z - info: sending command...
2020-05-03T16:01:16.508Z - info: sending command...
2020-05-03T16:01:16.510Z - info: 
2020-05-03T16:01:16.512Z - info: ------------------------------ 15ms
2020-05-03T16:01:16.512Z - info: parsing response...
2020-05-03T16:01:16.513Z - info: ControllerMpd::parseTrackInfo
2020-05-03T16:01:16.515Z - info: ControllerMpd::pushError
2020-05-03T16:01:16.516Z - info:  message=Cannot read property 'split' of undefined, stack=TypeError: Cannot read property 'split' of undefined
    at Promise._successFn (/volumio/app/plugins/music_service/mpd/index.js:294:72)
    at nextTickCallback (/volumio/node_modules/kew/kew.js:47:28)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
    at process._tickCallback (internal/process/next_tick.js:180:9)
2020-05-03T16:01:16.517Z - info: ------------------------------ 28ms
2020-05-03T16:01:17.102Z - info: ControllerMpd::sendMpdCommand add "http://192.168.178.71:8000/hifiberry.mp3"
2020-05-03T16:01:17.103Z - info: sending command...
2020-05-03T16:01:17.105Z - info: 
---------------------------- MPD announces system playlist update
2020-05-03T16:01:17.106Z - info: Ignoring MPD Status Update
2020-05-03T16:01:17.106Z - info: parsing response...
2020-05-03T16:01:17.107Z - info: CoreStateMachine::setConsumeUpdateService mpd
2020-05-03T16:01:17.108Z - info: ControllerMpd::sendMpdCommand play
2020-05-03T16:01:17.109Z - info: sending command...
2020-05-03T16:01:17.112Z - info: ------------------------------ 7ms
2020-05-03T16:01:17.113Z - info: parsing response...
```

I am using System version 2.729

xipmix, 2020-05-05:
Thanks for your report. I _think_ the error is coming from here (though the line numbers don't match up exactly)

https://github.com/volumio/Volumio2/blob/b2e85b2e8facd2d741f2fc40be426a34804b56fc/app/plugins/music_service/mpd/index.js#L269-L278

Somehow the 'trackType' field is not defined, which suggests parseTrackInfo  
https://github.com/volumio/Volumio2/blob/b2e85b2e8facd2d741f2fc40be426a34804b56fc/app/plugins/music_service/mpd/index.js#L365
 is mishandling this case.

I'm not sure what the fix is but it seems like if ```resp.isStreaming``` gets set True, perhaps ```resp.trackType = 'webradio'``` or something would be an appropriate thing to do.

The other odd thing here is the idiom being used. All the other interrogations of trackType would do it like this
```
collectedState.trackType = trackinfo.trackType.split('?').pop();
```

erols, 2020-05-06:
Just in case, I tried changing to 

`collectedState.trackType = trackinfo.trackType.split('?').pop();`
 
No luck.

Can I change log levels to get more information?

xipmix, 2020-05-06:
Yes that won't help if trackType is ```undefined```; it needs contain a string object to have that ```.split``` method.

I think the most helpful thing to do is try to dump the whole ```trackinfo``` object in the log.
It might be useful to uncomment this line in the ```parseTrackInfo()``` function
https://github.com/volumio/Volumio2/blob/b2e85b2e8facd2d741f2fc40be426a34804b56fc/app/plugins/music_service/mpd/index.js#L368

and try a similar line to dump the ```trackinfo``` object right before the line that is currently failing.

erols, 2020-05-09:
Hi, I updated Volumio in the meanwhile and now I don't volumio.log any more. I tried tailing mpd.conf with logging line uncommented but didn't see it report. Do I have to set a log level or enable volumio logs?

erols, 2020-05-09:
I got this output when I tried to play my station -
```
May 09 11:44:01 volumio volumio[910]: info: CoreCommandRouter::executeOnPlugin: webradio , handleBrowseUri
May 09 11:44:05 volumio volumio[910]: info: CoreCommandRouter::executeOnPlugin: webradio , handleBrowseUri
May 09 11:44:09 volumio volumio[910]: info: CoreCommandRouter::volumioAddQueueItems
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::addQueueItems
May 09 11:44:09 volumio volumio[910]: info: CorePlayQueue::addQueueItems
May 09 11:44:09 volumio volumio[910]: info: Adding Item to queue: 192.168.178.71:8000/hifiberry.mp3
May 09 11:44:09 volumio volumio[910]: info: CoreCommandRouter::executeOnPlugin: webradio , explodeUri
May 09 11:44:09 volumio volumio[910]: info: CorePlayQueue::saveQueue
May 09 11:44:09 volumio volumio[910]: info: CoreCommandRouter::volumioPushQueue
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::updateTrackBlock
May 09 11:44:09 volumio volumio[910]: info: CorePlayQueue::getTrackBlock
May 09 11:44:09 volumio volumio[910]: info: CoreCommandRouter::volumioPlay
May 09 11:44:09 volumio volumio[910]: UNSET VOLATILE
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::play index 18
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::setConsumeUpdateService undefined
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::stop
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::setConsumeUpdateService undefined
May 09 11:44:09 volumio volumio[910]: UNSET VOLATILE
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::updateTrackBlock
May 09 11:44:09 volumio volumio[910]: info: CorePlayQueue::getTrackBlock
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::stPlaybackTimer
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::pushState
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::getState
May 09 11:44:09 volumio volumio[910]: info: CorePlayQueue::getTrack 16
May 09 11:44:09 volumio volumio[910]: info: CoreCommandRouter::volumioPushState
May 09 11:44:09 volumio volumio[910]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::serviceStop
May 09 11:44:09 volumio volumio[910]: info: CorePlayQueue::getTrack 16
May 09 11:44:09 volumio volumio[910]: info: CoreCommandRouter::serviceStop
May 09 11:44:09 volumio volumio[910]: info: ControllerMpd::stop
May 09 11:44:09 volumio volumio[910]: info: ControllerMpd::sendMpdCommand stop
May 09 11:44:09 volumio volumio[910]: info: sending command...
May 09 11:44:09 volumio volumio[910]: info:
May 09 11:44:09 volumio volumio[910]: ---------------------------- MPD announces state update: player
May 09 11:44:09 volumio volumio[910]: info: parsing response...
May 09 11:44:09 volumio volumio[910]: info: ControllerMpd::getState
May 09 11:44:09 volumio volumio[910]: info: ControllerMpd::sendMpdCommand status
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::play index undefined
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::setConsumeUpdateService undefined
May 09 11:44:09 volumio volumio[910]: info: sending command...
May 09 11:44:09 volumio volumio[910]: info: CorePlayQueue::getTrack 18
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::startPlaybackTimer
May 09 11:44:09 volumio volumio[910]: info: CorePlayQueue::getTrack 18
May 09 11:44:09 volumio volumio[910]: info: [1589024649100] ControllerWebradio::clearAddPlayTrack
May 09 11:44:09 volumio volumio[910]: info: ControllerMpd::sendMpdCommand stop
May 09 11:44:09 volumio volumio[910]: info: sending command...
May 09 11:44:09 volumio volumio[910]: info: parsing response...
May 09 11:44:09 volumio volumio[910]: info: parsing response...
May 09 11:44:09 volumio volumio[910]: info: ControllerMpd::parseState
May 09 11:44:09 volumio volumio[910]: info: ControllerMpd::sendMpdCommand playlistinfo
May 09 11:44:09 volumio volumio[910]: info: ControllerMpd::sendMpdCommand clear
May 09 11:44:09 volumio volumio[910]: info: sending command...
May 09 11:44:09 volumio volumio[910]: info: sending command...
May 09 11:44:09 volumio volumio[910]: info:
May 09 11:44:09 volumio volumio[910]: ---------------------------- MPD announces system playlist update
May 09 11:44:09 volumio volumio[910]: info: Ignoring MPD Status Update
May 09 11:44:09 volumio volumio[910]: info: parsing response...
May 09 11:44:09 volumio volumio[910]: info: parsing response...
May 09 11:44:09 volumio volumio[910]: info: ControllerMpd::parseTrackInfo
May 09 11:44:09 volumio volumio[910]: info: ControllerMpd::sendMpdCommand load "192.168.178.71:8000/hifiberry.mp3"
May 09 11:44:09 volumio volumio[910]: info: ControllerMpd::pushState
May 09 11:44:09 volumio volumio[910]: info: CoreCommandRouter::servicePushState
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::syncState
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::pushState
May 09 11:44:09 volumio volumio[910]: info: CoreStateMachine::getState
May 09 11:44:09 volumio volumio[910]: info: CorePlayQueue::getTrack 18
May 09 11:44:09 volumio volumio[910]: info: CoreCommandRouter::volumioPushState
May 09 11:44:09 volumio volumio[910]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
May 09 11:44:09 volumio volumio[910]: info: CorePlayQueue::getTrack 18
[... cut ...] ping notification.Current webradio Received mpd
May 09 11:44:09 volumio volumio[910]: info: sending command...
May 09 11:44:09 volumio volumio[910]: info: ------------------------------ 69ms
May 09 11:44:09 volumio volumio[910]: info: null
May 09 11:44:09 volumio volumio[910]: info: ------------------------------ 44ms
May 09 11:44:09 volumio volumio[910]: info: ControllerMpd::sendMpdCommand add "192.168.178.71:8000/hifiberry.mp3"
May 09 11:44:09 volumio volumio[910]: info: sending command...
```

erols, 2020-05-12:
I missed off "http://" when I added the webradio in the Volumio UI and that was causing the problem.

closed 2020-05-12 by erols